opentx/opentx #7291: v2.3.4 Jumper T16: eu not respected !?

olliw42, 2020-01-12:
I've noted that in the fwoptions.py file, https://github.com/opentx/opentx/blob/2.3/radio/util/fwoptions.py, the options_jumper_t16 dictionary doesn't have a "eu" entry unlike e.g. for the horus X10 (which I understood should be the same firmwares)

So I've upgraded my jumper t16 today to v2.3.4, and had the eu flag set in Companion, the firmware file is named opentx-t16-eu-internalmulti-lua-ppmus-en-2.3.4.bin. (the VERSION screen says af69e36a)

In the OPTS in the VERSION info page eu is however not listed and the country code was not Europe but America. 

I'm not yet familiar enough with the whole OpenTx system, and thus am not sure if my interpretation is correct, but I interpret the results to confirm that eu is not respected for the T16.

kilrah, 2020-01-13:
eu is irrelevant for the t16, that's only effective for an xjt module. Should rather remove the companion option.

3djc, 2020-01-13:
maybe for external ?

olliw42, 2020-01-13:
thx

the shown BuildOptions checkboxes in Companion/EditSettings are clearly adaptive to the chosen RadioType, so, if eu is indeed irrelevant I would think it should not be offered in the Companion for the T16

not sure what the country code is then for but this is then clearly just my newbie status. If it however should happen that it is what one would have to set in order to make adjustments like hide/not hide the D8 choice, then I guess I would find it most natural to keep the eu and rather change the radio code to set that field, and whatever else is relevant, by firmware. This keeps "compatibility" and might bring advantages if it should happen that some eu/non-eu things come up later.

3djc, 2020-01-13:
eu option really shouln't even be there it serve no real purpose, but some like it for some reason. I'll remove it for t16

olliw42, 2020-01-13:
great
helps at least the unexperienced user to get confused :):)
thx

kilrah, 2020-01-13:
The eu option is mainly there for factory builds since radios sold in the EU should not have D8 accessible (not legal for sale anymore).
Some people want to be able to keep it that way when they upgrade so they can show their club reps that their radio is compliant...

Ceeb182, 2020-01-13:
I would like to use this thread to propose an idea.
Is it possible to make the "EU" compilation option prevent the selection of FCC protocols for both the internal and external module, no matter which HF module the transmitter is equipped with (XJT, iXJT, ISRM, R9M, Muliprotocol...) ?
Of course, OpenTx can only do this with HF modules it recognizes. So it is possible to know:
- either according to the radio type used (X9D, X7, X10, X12, T12, T16, TX16...) for the internal module
- or if an external module recognized by OpenTx is used: multiprotocol, R9M

3djc, 2020-01-13:
we don't have any way to recognize if a module is fcc or lbt

Ceeb182, 2020-01-13:
Oh, yeah. Sorry I didn't give it enough thought.
Indeed in "EU" there is the limitation for an R9M to 25mW and 500mW (not 1W) but it's impossible to know if the firmware of the R9M is EU-LBT or FCC. 
It is the same thing for an XJT or iXJT, in "EU" there is the limitation for a D16 (not D8) but impossible to know if the firmware of the XJT is EU-LBT or FCC.
For the ISRM, it's maybe different because it seems to me that it's possible to see the firmware version of the module.
For Multiprotocol, I don't know. It seems to me that OpenTx is talking to the module, but I'm not sure if the information is relevant enough to make a filter.

Anyway, it's not that simple. 
I completely understand that the OpenTx team has other priorities.

kilrah, 2020-01-13:
Seems very much pointless and a waste of time.

olliw42, 2020-01-16:
didn't go into 2.3.5, despite milestone, right?

3djc, 2020-01-16:
Right, which is why this is still oprn

PR #7295 (2020-01-30): Remove uneeded D8 option on some radio
This closes #7291
closed 2020-01-30 by bsongis