opentx/opentx #7292: selectable RF OFF state on powerup

i3dm, 2020-01-13:
Hello all,

when setting up a model memory, doing SW updates or using simulator functions, no RF link is required (neither external or internal module). therefore it would be useful to offer a selectable "RF OFF" state on powerup if chosen.
a regular powerup should always go to RF ON by default.

Futaba allows this when switching on the Tx with throttle above idle. after disabling the throttle alert, it asks you if you want to run RF ON or RF OFF. 
i think this is ectremely useful and would be a great feature to add to OpenTX.
thank you.

robustini, 2020-01-13:
Indeed it would be interesting.

Eldenroot, 2020-01-15:
Great idea!

i3dm, 2020-07-03:
Has this issue been implemented on 2.3.6? i see it was assigned to milestone 2.3.6 but havent seen anything about implementation?

3djc, 2020-07-03:
Still open so not implemented

i3dm, 2020-07-03:
which milestone will this go into?

3djc, 2020-07-03:
Undefined. Depends when someone decides to spend several hours off their free time to work on this rather than other open issues

i3dm, 2020-12-31:
any news about this feature request?

i3dm, 2021-06-30:
still think this is a great feature