opentx/opentx #7293: Selectable Trim step for throttle

i3dm, 2020-01-13:
Some types of models, such as turbine jets, require a high trip step for throttle (2-4 clicks to go from cut to idle trim).

this is currently only supported on OpenTx for all trims together, which is not useable as making coarse trim corrections for flight controls is not an option.
it would be great to either separate just the throttle, or even better - to enable a trip step for each trim individually.

thank you.

kilrah, 2020-01-13:
Using the trim for idle/cutoff is a remnant of when trims were mechanical, makes little sense to do it that way nowadays. You can use a switch, or even use the trim key as a switch that behaves whatever way you want...

closed 2020-01-13 by kilrah

i3dm, 2020-01-13:
using a switch is not safe enough.
using the trim key is a switch is what im doing now. but a huge amount of pilots i know still use the fuel trim range in 2 or 4 clicks (large steps) so i thought it could be a good feature.

3djc, 2020-01-13:
Once you get your mind into it, you can do really nice stuff with opentx. You could program it so that it needs say 4 clicks within a timeframe of 2 secs to be activated, for example

i3dm, 2020-01-13:
trust me my mind is into it and I've done some cool things. but in this case simplicity would win. that's my opinion at least.

i have heard from several pilots that they are a bit afraid of opentx due to the complexity to perform an otherwise simple task done in other radios. this is one of those cases.

craigywsm, 2022-01-09:
I agree that this feature should be re opened. 
you want the radio to inform you full trim and low trim. 
it would be a much better idea to allow the throttle to have its own trim setting. I belive this is being developed in Ethos. Please can this be developed in opentx?

i3dm, 2022-01-09:
This is not an opentx forum.
@bsongis can you please increase max trip step from 128 (alpha 15) to 256?
this would mean 2 clicks from cut to  idle for jets.

craigywsm, 2022-02-07:
please can this be re opened?

i3dm, 2022-02-07:
> please can this be re opened?

i dont think anyone is still working on Opentx...?

craigywsm, 2022-02-07:
really?

craigywsm, 2022-02-07:
@bsongis please chap?

bigtwin, 2022-02-08:
If you guys dont feel like anyone is still working on OTX and you are convinced that your feature request is very important and helpful to many others, why dont you give EdgeTX try? It is supposed to be more open to new features. May that development team implement your request.
Independend of your favorit, consider to donate!

i3dm, 2022-02-08:
youre probably right, opentx seems to be dead and would probably be a good idea to migrate to EdgeTx.
Can only hope its as stable as OTX 2.3.14

craigywsm, 2022-02-08:
@bigtwin I have and will continue to donate to OpenTX and have no interest in switching the EdgeTX. I fly jets on my X9D 2019SE and would like this feature as would make my life easier with the start sequence. If this could get developed I would be more than happy to pay for it. @bsongis

i3dm, 2022-02-08:
> @bigtwin I have and will continue to donate to OpenTX and have no interest in switching the EdgeTX. I fly jets on my X9D 2019SE and would like this feature as would make my life easier with the start sequence. If this could get developed I would be more than happy to pay for it. @bsongis

i may have some other options for you. email me please - i3dm@hotmail.com

jrwieland, 2022-02-08:
OpenTx is still being worked on.   

It is true that fewer developers are currently working on the issues and performing pull requests since EdgeTX split from the main development line of OpenTX;  but it is still active (check out the insights for the last 30 days to learn just how active).  Enhancements are being made both from the EdgeTX's code migrating in and from issues/pulls directly related to OpenTX.  

If it wasn't for the notifications I receive, I wouldn't know that this issue (buried on page 31 of the closed issues) was even back up for discussion the last couple of days after being closed for a year.  

I find that if I want a discussion on an item, it is better to discuss it on the OpenTX discord.  There you can get quicker feedback and the ability to tap into additional resources.  

my 2 cents.

3djc, 2022-02-08:
In this very case, we highlighted several alternative ways to achieve the feature, and we are not willing to add extra complexity for the other users

i3dm, 2022-02-08:
> In this very case, we highlighted several alternative ways to achieve the feature, and we are not willing to add extra complexity for the other users
which complexity would that be?

craigywsm, 2022-02-08:
> > In this very case, we highlighted several alternative ways to achieve the feature, and we are not willing to add extra complexity for the other users
> > which complexity would that be?

I agree this feature would be an advantage not adding complexity. 
complexity is the work around with mixes that could compremise a jet worth thousands.
As a software support manager myself I am discussed in some of the comments on this ticket! especially as I donate to this project on technical advice and financial