opentx/opentx #7298: [OpenTX V2.3.5] - Compilation issue, how fix it?

robustini, 2020-01-15:
I need some help, i've this error when compiling 2.3.5:

Scanning dependencies of target datacopy
[  6%] Generating datacopy.cpp
  File "/home/marco/opentx/radio/util/generate_datacopy.py", line 50
    print("Warning more than one union member (%s) in anynomous union inside struct %s, consider NOBACKUP statements" % (uc.spelling, cursor.spelling), file=sys.stderr)
                                                                                                                                                            ^
SyntaxError: invalid syntax
radio/src/CMakeFiles/datacopy.dir/build.make:61: recipe for target 'radio/src/datacopy.cpp' failed
make[3]: *** [radio/src/datacopy.cpp] Error 1
make[3]: *** Deleting file 'radio/src/datacopy.cpp'
CMakeFiles/Makefile2:239: recipe for target 'radio/src/CMakeFiles/datacopy.dir/all' failed
make[2]: *** [radio/src/CMakeFiles/datacopy.dir/all] Error 2
CMakeFiles/Makefile2:314: recipe for target 'radio/src/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [radio/src/CMakeFiles/firmware.dir/rule] Error 2
Makefile:264: recipe for target 'firmware' failed
make: *** [firmware] Error 2

robustini, 2020-01-15:
CMake Deprecation Warning at companion/src/CMakeLists.txt:358 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

bsongis, 2020-01-15:
Are you using python2?

robustini, 2020-01-15:
Yes, V2.7.17.

bsongis, 2020-01-15:
python3 is needed, not python2

closed 2020-01-15 by bsongis

robustini, 2020-01-16:
@bsongis please could you put me in the right direction?

python --version
Python 3.6.9

But:

[image]

robustini, 2020-01-16:
Yes, I have omitted this step, I try, thanks.

robustini, 2020-01-16:
Now:

-- Found Git: /usr/bin/git (found version "2.17.1") 
-- OpenTX 2.3.6 @cc298988
-- Found PythonInterp: /usr/bin/python3.6 (found version "3.6.9") 
-- Python found, version: 3.6.9
-- Qt Version: 5.9.5

But:

-- Found Git: /usr/bin/git (found version "2.17.1") 
-- OpenTX 2.3.6 @cc298988
-- Found PythonInterp: /usr/bin/python3.6 (found version "3.6.9") 
-- Python found, version: 3.6.9
-- Qt Version: 5.9.5

Where am I wrong?

robustini, 2020-01-16:
update-alternatives --list python

/usr/bin/python2.7
/usr/bin/python3.6

python --version

Python 3.6.9

robustini, 2020-01-16:
I removed the alternative for version 2.7, now there is only 3.6, but I have the same error.

update-alternatives --list python

There is only one alternative in the python group (which provides / usr / bin / python): /usr/bin/python3.6
Nothing to configure.

bsongis, 2020-01-16:
pip install clang ?

robustini, 2020-01-16:
Yes, missing clang for Python 3, thanks!

bsongis, 2020-01-16:
would you update our wiki page?

robustini, 2020-01-16:
It would be better if someone who perfectly knows the English language did it, this is not my case.

robustini, 2020-10-03:
I'm sorry @bsongis if I still bother you in a closed issue but for a few days I can't compile in WLS anymore (updated Ubuntu 18), I've this error:

[image]

3djc, 2020-10-03:
thats the wrong compiler version

robustini, 2020-10-03:
> 
> 
> thats the wrong compiler version

Ok, what should I do?

kilrah, 2020-10-03:
As mentioned in the wiki compiling for radio requires an old version, 4.7.
No idea how to install that on WSL.

robustini, 2020-10-03:
> 
> 
> As mentioned in the wiki compiling for radio requires an old version, 4.7.
> No idea how to install that on WSL.

The problem is that until a few days ago I was compiling without issue, and I haven't updated anything.

[image]

kilrah, 2020-10-03:
This is code that was written a week ago, and apparently it doesn't make that compiler version happy...

robustini, 2020-10-03:
Same issue with g++4.8, damn!

[image]

robustini, 2020-10-03:
The problem is that I can't find g ++ 4.7 arm64 for Ubuntu 18, starts from version 4.8 which still gives that problem.
Wouldn't it be better to check the code?

3djc, 2020-10-03:
No, compiling with any other version than the tested compiler is really, not something anyone should be doing

kilrah, 2020-10-03:
Can you pull and try again?

robustini, 2020-10-03:
> 
> 
> Can you pull and try again?

Fixed, thanks!!!