opentx/opentx #7299: OpenTx 2.3.5 doesnt run X10S express

i3dm, 2020-01-15:
X10S Express, after upgrading to 2.3.5 got "Wrong PCB Deteced".
tried 2 different compliation versions - same results.
thank you.

bsongis, 2020-01-15:
Confirmed

bsongis, 2020-01-15:
2.3.4 was OK? I am comparing changes ...

i3dm, 2020-01-15:
Yes.

i3dm, 2020-01-15:
can you complite 2.3.4 for me and send it over temporarily so my radio isnt bricked? :)

3djc, 2020-01-15:
2.3.4 should not be used on horus and t16 anymore

i3dm, 2020-01-15:
im aware of that, was there an issue with using internal module ONLY?

bsongis, 2020-01-15:
I see the issue.

i3dm, 2020-01-15:
is 2.3.6 coming?

closed 2020-01-15 by bsongis

i3dm, 2020-01-15:
how do i get the fix?

bsongis, 2020-01-15:
wait one hour, we will patch the server directly to avoid an extra delay

i3dm, 2020-01-15:
do you rewrite 2.3.5 or is that 2.3.6?

3djc, 2020-01-15:
2.3.5

i3dm, 2020-01-15:
thanks guys. how do i tell when this version will be online?

dev-fred, 2020-01-15:
👍

bsongis, 2020-01-15:
it's online now

Ghost-B3, 2020-03-15:
It´s not working for me too. I´m running EU LBT version.
I´m still getting "wrong PCB Detected" message after flashing my radio with 2.3.5 version.

commit 4d5b5e1c (2020-01-15): Fixes #7299
commit 4fbe404c (2020-03-15): Fixes #7299