opentx/opentx #7303: Missing units for telemetry values

MarkusPfaff, 2020-01-16:
Hi,
Currently I'm working on a sensor that is capable of measuring the inner resistance of the flight battery cells and results are rather prommissing already. Only there is no unit Ohm (or better suited in this case mOhm) which would be needed for that purpose. Also there is no sensor value ID for a resistance value -- Telemetry value "Id" in companion lingo is meant, not the physical sensor unit number called "Instance" by the companion. 

Any chance to add that unit and an appropriate ID to the OpenTx portfolio?

The same for the unit seconds (sec, ms, us). Since my sensor adds time measurement (which cannot be done with the transmitter timer in the needed exact way) that unit would also be needed.

Having worked a lot with OpentTx telemetry a question comes up: Why are the sensor value IDs for telemetry entangled so intensely with FrSky product offerings? This way there are n different IDs for temperature values for example. These IDs differ mainly in the specific FrSky product they are used in -- although they in fact might behave differently in an undocumented way, because of range limits associated with these products. Example? ESC temperature (ESC_TEMPERATURE_FIRST_ID) vs. gas suite temperature (GASSUIT_TEMP1_FIRST_ID). Some IDs also fork up into two values in transmitter telemetry (RBOX_CNSP_FIRST_ID).

A sensor unit from outside FrSky is forced to pick its value IDs from several of the FrSky sensor products, often in a way that looks like a total mess (e.g. OpenXsensor, SM Modellbau). OpenTx is still not a FrSky product, or is it?

I'd propose to let FrSky have its sensor ID chaos as it is, but OpenTx should give the community a set of IDs that are only related to physics instead of products. The range of IDs is large enough at 16 bits and easily provides this. The DIY_FIRST_ID (for Do It Yourself I suppose) looks already a bit like this, but carries no units.
Of course FrSky may or may not use these physical IDs on future products the same way as any other supplier or community member, but the right to define this belongs to OpenTx in my opinion not to FrSky.

Best Regards,
Markus

3djc, 2020-01-16:
In this case, it is sport telemetry protocol, which is FrSky product. IDs and such are different if you use Hitec, for Hott telemetry for exemple.

kilrah, 2020-01-16:
Becasue it's how it is. The sport protocol is frsky proprietary, even if everyone has been hooking onto it, it's their specs. 
So if you make a sport sensor that's what you're constrained to.

That said the DIY range is precisely one that FrSky have said they would not use so that people could use the large number of IDs in it for whatever they want. But obviously there's no standard there and nobody has taken on the job of defining it.

MarkusPfaff, 2020-01-16:
The spec is for the way the data gets from the sensor to the receiver and from there to the transmitter. The complete rest of the way - and a rather long part of it - is handled by OpenTx.  Personally, I simply don't see why FrSky should be granted the sole right to dictate the telemetry value usage on the OpenTx community which at the same time provides a rather large part of the overall functionality (which keeps money rolling in the FrSky front door), but the OpenTx community gets so little influence in return. Simply feels unfair in my opinion. 

Maybe a way to go would be to propose a meta definition which sits on top of sport, Hitec, Hott,... in an attempt to make things less vendor independent.

All those more or less philosophical questions still leave open the question about the units mOhm and sec, ms, us. These would be pure OpenTx definitions for which no telemetry protocol definition is in the way. Maybe they could be added to OpenTx to make it even more useful than it already is?

bsongis, 2020-01-16:
I think it's not complex, just add an unit in place of UNIT_SPARE1

bsongis, 2020-01-16:
And don't forget to update UNIT_MAX

MarkusPfaff, 2020-01-16:
Bertrand,

would be great if that could be integrated in a future OpenTx revision.

Just got news from the German FrSky supplier about the solution of the "Great Bug" in ACCST D16: 
https://www.frsky-rc.com/important-firmware-update-accst-d16/

Perhaps it would be a good idea to drop a note on the OpenTx homepage for those who are not aware of it.

Thanks
Markus

MarkusPfaff, 2020-07-14:
I'd still be interested in the simple addition of missing units.

MarkusPfaff, 2021-01-11:
Still interested in making our telemetry more universal.

MarkusPfaff, 2021-05-22:
Again, How about enabling (future) work on sensors by adding units that are beyond the imagination of FrSky developers? Just add the units.

- Milliohms (e.g. for inner resistance of batteries)
- Ohms
- Mikroseconds (e.g. for the PWM setting of a servo which may be interesting to get some feedback from the Rx on what it gets from the Tx)
- Milliseconds (e.g. for time measurements using KI-based sensors)
- Seconds
- all length units (e.g. for LIDAR and RADAR sensors)
- light intensity units

MarkusPfaff, 2022-04-17:
The issue still persists.