opentx/opentx #7307: Maybe an error with rx and TX 2.0.1 and telemetry

andreasstamm, 2020-01-19:
On the germen frsky forum there is the discussion of there is maybe an bug with the telemetry and /or sxr config Skript and the fw 2.0.1. And opentx 2.3.5.
Tx used for test is x9dp, X10S and x12s

On x9dp and x10s there occurring sometimes missvalues on the telemetry. 
The sxr script need some seconds to load data on the sxr rx and the often misses some values displayed as - - -. 

I don't k ow if it is an 2.0.1 problem. 
Greetings

3djc, 2020-01-19:
Please conduct the same test with a different version of OpenTX, but with d16 v2.

andreasstamm, 2020-01-19:
I had a x7s and flashes it back.. But hangs up when I try to switch page2 on the s6r

3djc, 2020-01-19:
So here you go, issue is not OpenTX. Closing this, please reopen if you find evidence that OpenTX is part of the problem

closed 2020-01-19 by 3djc

kilrah, 2020-01-19:
Several people have done test and condluded that D16 V2 for the S6R/S8R was broken in many ways.

andreasstamm, 2020-01-19:
Hi kilrah. . Many people talk about not only script problems with the s6r and S8R. And with this rx some telemetry values jumping some times on high error values. FYI if have tested the x7s with 2.3.5 and the v2.0.1 with an g-rx8 and the error is with the g-rx8 too. So maybe this script problem occur because the telemetrie channel gets error values
Greetings Andreas