opentx/opentx #7308: Dockerfile is missing from radio/utils

blairfancy, 2020-01-19:
Compilation wiki guide recomands using docker for relaible build of opentx but radio/utils/Dockerfile 
does not exist and 
command 
`ocker build -t opentx .`

return:
`unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/blairfancy/source/OpenTX_off_2.3/opentx/radio/util/Dockerfile: no such file or directory`

**General information**
 - OpenTX version 2.3 branch of git
 - Hardware not applicable
 - OpenTX Companion not applicable

vitas, 2020-01-24:
i was also surprised why it was removed..
use my image to build firmware
https://hub.docker.com/r/vitass/opentx-fw-build

gretel, 2020-03-06:
any update on this?

richardclli, 2020-08-04:
Why docker file removed?

richardclli, 2020-08-04:
@vitas your docker image has bug when trying to build t18 binaries, your scripts has some problem, I fixed it as below:

elif board_name == "t18":
    extra_options["PCB"] = "X10"
    extra_options["PCBREV"] = "T18"
    firmware_options = options_jumper_t18
    maxsize = 2 * 1024 * 1024

gretel, 2020-08-04:
this feels like closed source to me.

vitas, 2020-08-04:
> @vitas your docker image has bug when trying to build t18 binaries, your scripts has some problem, I fixed it as below:
> 
> elif board_name == "t18":
> extra_options["PCB"] = "X10"
> extra_options["PCBREV"] = "T18"
> firmware_options = options_jumper_t18
> maxsize = 2 * 1024 * 1024

that string is presented, not clear what you have fixed
https://github.com/vitas/opentx-docker-build/blob/master/build-fw.py

richardclli, 2020-08-05:
@vitas Your script has a typo, you used cmake_options in T18, you should use extra_options instead.

richardclli, 2020-08-10:
I have checked the log, it is 3djc who remove the Dockerfile, with the reason of housekeeping on 11/13/2019.  Not sure if he accidentally delete the file or having other intention.

@3djc Any comments?

3djc, 2020-08-10:
It is no accident. This file is not required to build OpenTX, we have no use for it, and do not whish to maintain it

closed 2020-08-10 by 3djc