opentx/opentx #7310: crossfireTelemetryPush function cause emergency mode on T16

lichtl, 2020-01-19:
**Describe the bug**
crossfireTelemetryPush function cause to crash openTX 2.3.5 (T16 will go into emergency mode / X9D will not properly execute function) when send 23 byte.

**To Reproduce**
Steps to reproduce the behavior:
1. Add/setup model with external Crossfire (no module required)
2. Add following lua on the SD card

```
local function run(event)
    lcd.clear()
	if event == EVT_ENTER_LONG then
        lcd.drawText(10, 20, "23 byte - PID change", 0)
        local result = crossfireTelemetryPush(0x78, {234,200,67,18,11,184,0,35,39,16,11,184,0,35,39,16,31,64,0,50,19,136,135})
        lcd.drawText(10, 35, "done", 0)
    end
	return 0
end

return { run=run }
```
3. Launch lua and hold ENTER key to trigger (T16 will go into emergency mode / X9D will not send out command) - https://youtu.be/ImHcW7YLh1A

**Expected behavior**
Command should be executed and send to Crossfire module

**General information**
 - OpenTX version (2.3.4 and 2.3.5 tested)
 - Hardware (Frsky X9D+, Jumper T16 both tested.)

3djc, 2020-01-20:
Anything above 16 will get you into trouble I think. Will double check with TBS

fedorcomander, 2020-01-20:
TBS packet is up to 64 bytes. It was on OTX 2.2.

3djc, 2020-01-20:
If you where sending 64 bytes in 2.2, you where probably massively exceeding buffer size (#define TELEMETRY_OUTPUT_FIFO_SIZE 16)

fedorcomander, 2020-01-20:
Great… Well… There are specs for crossfire protocol, and buffer must be extended to 64 bytes. Sorry.

3djc, 2020-01-20:
Not sure we have the memory available for that, we will have a look, not commiting to that

fedorcomander, 2020-01-20:
Well, telemetry receiving buffer is 256 bytes if i am not mistaken. which is quite a lot. can be smaller? ;)

kilrah, 2020-01-20:
We can't control when we receive stuff, but we can control when we send stuff. Normal to have RX bigger.

fedorcomander, 2020-01-20:
Yes, thats true. But specs are specs. 

Meanwhile, i patched device.lua so it will work with float numbers that kiss uses a lot. That DOESNT let you guys off the hook with buffer size ;)

syrekszymon, 2020-01-21:
> ...Meanwhile, i patched device.lua ...

Where can i find it?

AngusCox, 2020-01-21:
> > ...Meanwhile, i patched device.lua ...
> 
> Where can i find it?

Here, https://www.patreon.com/posts/33299763

3djc, 2020-01-21:
We will extend the buffer to 64 and put safeguards

fedorcomander, 2020-01-22:
perfect! thank you from all kiss users.

PR #7323 (2020-01-30): Increase lua telemetry push output buffer
- Increase size to 64 bytes as per Crossfire specs. 
- Protect against overflow

This fixes #7310
closed 2020-01-30 by bsongis