opentx/opentx #7319: lua does not execute from Special Function using "one" switch

sande005, 2020-01-21:
- OpenTX version 2.3.5
 - Hardware Frsky X9D+ SE
 - OpenTX Companion 2.3.5 on Win 10

In Companion Simulator **and** on Radio:  Lua function script generates "xx.luac" file, but does not execute on startup, when invoked from a Special Function using a "one" switch.

In Companion Simulator, if the "xx.lua" file is deleted (but not the "xx.luac" file), it then does run on Simulator startup.  On the radio, it will not run from just the "xx.luac" file, as expected.

Script does run as expected, if manually started by navigating to the SD card, in both Simulator and on Radio.

Attached script compares today's day-date to GV1.  If no match, it updates GV1 to today's day-date, resets GV2 to 0, and resets timer 2.

"luatest.otx" renamed to "luatest.txt"
[luatest.txt](https://github.com/opentx/opentx/files/4092262/luatest.txt)
"luatst.lua" renamed to "luatst.txt"
[luatst.txt](https://github.com/opentx/opentx/files/4092278/luatst.txt)

sande005, 2020-01-22:
The lua script ran fine in Companion 2.2.4.  Uninstalled/reinstalled 2.3.5 and still no go.
Added some print lines, to aid in looking at the debug output.  Attached is the comparison.  2.2.4 all good, 2.3.5 no output.

[debug.txt](https://github.com/opentx/opentx/files/4095525/debug.txt)

sande005, 2020-01-22:
Geez, I just give up!
I tested above in Companion 2.2.4, and it worked fine.
Went to the work of downgrading both radios back to 2.2.4.  Yea backups!

But...while it still works in Companion...no go on the radio when initiated from a "one" special function.
Both on the X9D+ and the X9D.

Since this is a gas plane, I now just have it run when I pull the ignition switch (SH)...where it behaves perfectly.....

jfrickmann, 2020-01-31:
"ONE": script only runs one time on startup
"ON": script runs continuously
Right?

kilrah, 2020-01-31:
Yes

closed 2020-10-28 by stale[bot]