opentx/opentx #7320: OpenTX 2.3.5 setTelemetryValue() doesn't work

lshems, 2020-01-21:
Am I doing something wrong? 

`model.insertInput(1-1, 1-1, {name="LUA"})
setTelemetryValue(0x0001, 0, 1, 0, 0, 0, 'LUA')`

run as one time does insert an inputline, but does not create a new sensor with name LUA.

??

3djc, 2020-01-21:
How exactly are you detecting new sensors while running a one time script ?

lshems, 2020-01-21:
Ok, so the setTelemetryValue does only create the telemetry value, and by detecting the sensors, it gets created. 
Clear.

Then the documentation is out of date and based on 2.1 even in the 2.2 lua guide. In 2.1 you don't have to detect sensors, so executing this command automatically adds the sensor.

Closing this, or will there be a doc update for 2.3 LUA??

3djc, 2020-01-21:
No that hasn’t changed, was always like that I believe

lshems, 2020-01-21:
Ah, then the documentation could be clearer :). It suggest you add a sensor by first use of the function

lshems, 2020-01-21:
Still some problems. I can discover nicely, using for instance the settings from the RSSI in the setTelemetryValue(). But it never seems to get the values from the simulator telemetry engine.

I cannot enter the instance in opentx. Is this used in some way.

lshems, 2020-01-21:
Well, seems it is working as intended.

closed 2020-01-21 by lshems