opentx/opentx #7321: Play Value Plays Starting Value Instead Of Ending Value

geekerator, 2020-01-22:
Please refer to the attachment. With openTX 2.3.4 on a Taranis X9E, I get what appears to be incorrect values when trying to play the value of CH9 (controlled by F1 and the curve CV1) when it changes by more than 5. The value played is the value BEFORE F1 was moved. Since |d(CH9)| > 5 cannot be be detected until F1+CV1 has caused CH9 to change by more than 5 (and hence L10 to be raised), shouldn't the ending value be played? As a side note, the same behavior occurs in Companion's simulator if F1 is ROTATED SMOOTHLY. However, ENDING values are played if I CLICK ON RANDOM POINTS around the circumference of F1. Any help would be appreciated. Thank you...

[PlayValueIssue.pdf](https://github.com/opentx/opentx/files/4095109/PlayValueIssue.pdf)

kilrah, 2020-01-22:
Please supply an otx, not a print.

geekerator, 2020-01-25:
Here is the .otx...

geekerator, 2020-01-25:
Posting an .otx file here was rejected. I attached the .otx to my reply to Andre's direct email (reply sent to: Andre Bernet <notifications@github.com>). If this is not correct, please let me know and I will comply...Jeff

kilrah, 2020-01-25:
Upload it somewhere anf post the link, e.g. https://www.filehosting.org/

geekerator, 2020-01-25:
Done. Here is the link:

https://drive.google.com/open?id=1M57S3DyEvbUV4E7qWuuQ0RlCZf7682ft

geekerator, 2020-01-25:
[](url)

geekerator, 2020-01-25:
Accidently closed.....

geekerator, 2020-01-27:
An additional clue: if the ("play value" of CH9) special function is initiated by an actual switch, the current value of CH9 is played. If the ("play value" of CH9) is initiated by a logical switch, it plays the starting value.

kilrah, 2020-01-27:
It's not the value before, it's the next value that is more than 5 off. Since LSs are evaluated >100 times per second even if you move the pot say +9%, during your move it will at some point be 5% and the play will be triggered there. Then since you're still moving, by the time the function gets executed it'll probably have moved a bit more so what gets played is maybe when you're at +7%.

closed 2020-01-27 by kilrah