opentx/opentx #7324: setTelemetryValue(id    , subID, instance, value, unit , precision, name  ) doesn't allow to set SCALE for voltage sensors

lshems, 2020-01-22:
The lua setTelemetryValue(id    , subID, instance, value, unit , precision, name  ) doesn't allow the scale to be set for voltage sensors, rendering the function practically useless for those sensors.

It would be very nice if that could be added as option.

getTelemetrySensor() would also be not bad as a function to manage the sensors better

3djc, 2020-01-22:
It is not meant to create sensors, but to simulate a flow of sensor data from lua data. If you want to scale the data, do it on lua side

lshems, 2020-01-22:
The docs says differently.

It would be nice to have the default scale 13.2 added at least when adding a voltage sensor.

lshems, 2020-01-22:
You cant prescale data when the scale on the sensor is autoset to zero without possibility to adapt from lua.

kilrah, 2020-01-23:
The 13.2 ratio is irrelevant to anything other than the Ax sensors. Use other IDs.

lshems, 2020-02-14:
No, rxbatt also.

Better would be if we would have real access to sensor definitions, but this could be a quick fix.

closed 2020-11-11 by stale[bot]