opentx/opentx #7326: New Wiki Page - Compiling OpenTX 2.3 on Windows

benlye, 2020-01-23:
A few of us who compile OpenTX on Windows have found that the instructions for configuring a Windows build environment were incomplete or unnecessarily complex, and didn't work well for OpenTX 2.3.  

We've re-written the instructions in a new Wiki page, which is currently here:
https://github.com/benlye/opentx/wiki/OpenTX-2.3-Windows-Build-Instructions

It would probably be helpful to any other users who want to compile on Windows to include these instructions in the main OpenTX wiki.

(It's not possible to create a pull request against the wiki, which is why I've created an Issue - I hope that's OK.)

[OpenTX 2.3 Windows Build Instructions.md.txt](https://github.com/opentx/opentx/files/4102552/OpenTX.2.3.Windows.Build.Instructions.md.txt)

3djc, 2020-01-23:
Have you confirmed emergency actually works correctly on your horus firmware (ie you still have control in case of emergency) ?

benlye, 2020-01-23:
No, I didn't know I needed to do that.  Is it a known issue with firmware compiled on Windows?  I searched but couldn't find any pointers.  How would I test it?

On the other hand, I'd be happy to change the page to add whatever disclaimer is needed if compiling firmware on Windows is generally a bad idea.

3djc, 2020-01-23:
We got biten ourselves recently, hence my asking ! To test, you need to set -DDEBUG=YES, and edit functions.cpp L38, uncomment the while(1). 

Compile a firmware, flash it, create a special function with 'test' function. Activating that SF will reboot the radio in emergency. You need to check that you still have servo mouvement at that stage (please do NOT test with a real model, just a receiver and a servo.

Be EXTREMELY carrefull not to fly with this and undo all those changes right after your tests

benlye, 2020-01-23:
Yikes!  Thanks for the extra info - I'll test it tonight and report back.

MRC3742, 2020-01-23:
I have helped benlye tweak his build instructions Wiki for OpenTX 2.3 as listed on his repo.

I'm building using Windows7 32bit and believe he is using Windows 10 64bit. I uninstalled all my previous build programs and files and then installed using his build instructions. We had to make a few changes for Win7 32bit and also as the build process recently changed requiring Python3.

Testing on Jumper T16, Internal MPM, FrSky D16 bound with (Ch1-8 Telem ON), FrSky X8R RX with eight servos attached. I have tested the Windows built firmware using the method described above by 3djc and have set up the Special Function using switch SA-up to test.

When SA-up is on LCD goes blank and X8R LED turns red causing no servo control. Switching to SA-down causes LCD to show "EMERGENY MODE" and X8R LED goes back to green. All eight servos are now operable in this state. Holding the Power Button to shut off from Emergency Mode states Receiver is still connected and requires "Select" to shut off.

This appears to work as expected? but I'm sure @benlye will test his build also.
@3djc Please Review and note if any additional testing is required

Rich

3djc, 2020-01-23:
Thats looks ok.

benlye, 2020-01-23:
I confirm - tested with a D8 receiver and a Betaflight flight controller so that I could observe all eight channels.

Loss of signal/control for ~120ms when emergency mode is triggered, then full control returns.

3djc, 2020-02-03:
Pushed into wiki, thanks Benlye :)

closed 2020-02-03 by 3djc