opentx/opentx #7328: Problème Télémétrie / simulateur Compagnon 2.3.5 (à partir de v2.3.0)

Jeep1656, 2020-01-23:
Sur simulateur Companion v2.3.5 (et à partir de la v2.3.0), les données ne semblent pas transmises vers la radio simulée:
 - en chargeant manuellement les valeurs (ex: altitude) dans la table.
- à partir d'un Log ouvert, ou les données arrivent bien dans la table.
Dans les 2 cas les pages télémétrie de la radio simulée restent vides et NO DATA clignote.
On remarque aussi que les inters logiques programmés à partir de valeurs de télémétrie (ex: altitude), ne réagissent pas non plus.
La case "Simuler" est bien cochée, les 2 autres cases n'agissent pas sur le problème.
Remarque 1: en réel, avec le modèle sous tension, la radio fonctionne parfaitement en Télémétrie.
Remarque 2: un retour fait en version Companion 2.2.4 a démontré que la télémétrie fonctionne alors parfaitement sur le Simulateur Companion 2.2.4.
Remarque 3: en V2.2.4 il n'y avait qu'une seule casse "Simuler" les 2 autres ayant été ajoutées sur la V2.3.0.
J'espère que les informations fournies vous permettrons de diagnostiquer le problème, je joins  2 copies d"ecrans.
merci
Jeep1656
[image]
[image]

3djc, 2020-01-23:
Please use English only please. Duplicate #6862

closed 2020-01-23 by 3djc

Jeep1656, 2020-01-24:
Please excuse my English: I use Google Translate to assist me.

 On the Companion v2.3.5 simulator (and from v2.3.0), the data does not seem to be transmitted to the simulated radio:
- 1/: By manually loading the values ​​(example: altitude) into the table.
- 2/: From an open Log, where the data arrives well in the table.
- 3/: In both cases the telemetry pages on the simulated radio remain empty and NO DATA flashes.
I also notice that the logical switches programmed with telemetry values ​​(example: altitude) do not react either.
The "Simulate" box is well checked, but the 2 other boxes do not act on the problem.
Note 1: in real life, with the model powered up, the radio works perfectly in Telemetry.
Note 2: I have tested with Companion 2.2.4, and then the telemetry works perfectly on the Companion 2.2.4 Simulator.
Note 3: in V2.2.4 there was only one case "Simulate" the other 2 having been added on V2.3.0.
I understood in other item on this site, that the Companion 2.3.0 simulator works well on Horus.
I have an X9E and a QX7 both in LBT-EU, OpenTX v2.3.5, the symptoms described are identical.
I hope that the information provided will allow you to diagnose the problem.
I enclose 2 copies of screens.
thank you
Jeep1656

Jeep1656, 2020-01-24:
Hello,
Excuse me, please, for this unjustified intervention.
I went directly to the submission of my problem, without knowing how your GitHub site works.
  I could have found the answer and how to avoid placing my intervention in the wrong place.
So I unsubscribed as you suggested.
Through the French Taranis forum, I know the work that your team provides.
I therefore wait, the situation not being blocking for me.
I am a fan of openTX, which I have only known for 6 months, but already converted!
Congratulations to you all.
jeep1656
PS: because of my bad English, I use Google Translate to help me. thanks for your understanding.

From: 3djc 
Sent: Thursday, January 23, 2020 8:26 PM
To: opentx/opentx 
Cc: Jeep1656 ; Author 
Subject: Re: [opentx/opentx] Problème Télémétrie / simulateur Compagnon 2.3.5 (à partir de v2.3.0) (#7328)

Closed #7328.

gastolectric, 2020-01-30:
Jeep1656,

Refer to my post on #6862 for at least a partial remedy to this issue - it should get most of your sensors working again

Jeep1656, 2020-01-30:
thank you,
the partial solution which is proposed is identical to that which I have already found. this makes it possible to operate the telemetry of the simulator
cordially

From: gastolectric 
Sent: Thursday, January 30, 2020 7:22 AM
To: opentx/opentx 
Cc: Jeep1656 ; Comment 
Subject: Re: [opentx/opentx] Problème Télémétrie / simulateur Compagnon 2.3.5 (à partir de v2.3.0) (#7328)

Jeep1656,

Refer to my post on #6862 for at least a partial remedy to this issue - it should get most of your sensors working again

robustini, 2020-06-03:
I confirm that there is this problem, I am doing tests and the real sensors of the model are not assigned to the telemetry of the simulator.
To solve I deleted all the real sensors in the sim and made a new scan, but in this way all the real model sensors are lost.
I hope to be fixed soon.

Jeep1656, 2020-06-03:
Hello,
Thank you for your intervention.
However, I recently switched to V2.3.7 and found that I had no more problems.
cordially