opentx/opentx #7329: Horus X12S 2.3.5 Problems updating firmware on X8R receivers

Luznatural, 2020-01-23:
I tried, with my X12S with the newest Open TX 2.3.5, also with IXJT internal module with firmware v2.01 (FCC) to update all my X8R receivers (total of 21 receivers) also to the newest firmware v2.01. Of course with the cable and the correct procedure I have been always using to do so (from the lower connectors in the external module bay, as the X12S doesn't have an auxiliary connector as the X10S).
Two different problems:
- If you try updating the firmware on the receiver directly in your airplane without disconnecting the servos, the process starts correctly but after a third of the way, the servos make a little jitter and the Horus tells you that is unable to reconnect with the receiver. The receiver does not block at all, as you can try it again and again with same results.
- If you don't have anything connected to the receiver, the process starts and run smoothly all the way to the end, but you are unable to bind it.

I ended up updating without problems all the receivers to v2.01 (FCC) with my reliable X9D+, also updated with Open TX 2.3.5 and the internal XJT module with firmware v2.01. Of course, I used the same cable and the same method. This time, I was perfectly able to update and bind with everything connected (servos, battery and sensors) to the receiver.

**General information**
 - OpenTX and Companion version 2.3.5
 - Hardware Frsky X12S and X9D+ (both with Open TX 2.3.5, and IXJT/XJT v2.01 firmware)
 - OpenTX Companion (2.3.5 on Windows 10 64bits)

3djc, 2020-01-23:
Please contact FrSky about those things, they ate not OpenTX related

Luznatural, 2020-01-23:
I do not think is related to FrSky at all, as I was able to do it with my Taranis. You have to remind that the capacity to programa external items from Horus was introduced very recently. I wouldn't close it so fast.

Luznatural, 2020-01-23:
Respectfully thanks in advance...

closed 2020-10-20 by stale[bot]