opentx/opentx #7330: Horus X12S: telemetry data messed up in Companion

Oldenport, 2020-01-23:
There seems to be an issue with Companion reading data from the Horus X12S: If I use the "read models and settings from radio" the download process does finishes without error. However if I simulate the radio in Companion the telemetry data in the widgets got messed up.
Example: First widget in the status bar is the value "Batt" on the transmitter. After reading the model data into Companion the same widget has the value "GV8". 
If I write back the model data via Companion the messed values get copied to the transmitter.

Steps to reproduce the behavior:
1) Open Companion
2) Connect transmitter via USB
3) go to "read models and settings from radio"
4) "simulate" radio in Companion
5) telemetry input value does not fit to transmitter settings anymore. eg. if the telemetry input value was "Batt" before it is "GV8" in Companion.

**General information**
 - OpenTX 2.3.5
 - FrSky Horus X12S
 - OpenTX Companion 2.3.5 (clean install after deleting older versions, including registry patch)
 - Win10

Oldenport, 2020-01-25:
I did another test today:
I deleted the radio and models folder from the transmitter. That automatically creates a new model to which I binded a receiver. I added the telemetry values to some widgets on the main screen and in the status bar. After that I imported the model data via Companion again. This time the telemetry values in the widgets did not change.
Therefore I assume that old/existing model data gets confused when reading into (new version of) Companion.
Am I the only one with this issue? Anything I can do to work around and not creating all models from scratch?

kilrah, 2020-01-25:
Never heard of anything like this. But then again we don't know where you come from before loading 2.3.5, so it's hard to make any assumptions.

Oldenport, 2020-01-25:
Sorry! Very valid point... I've update from 2.3.4. I also took a few steps before, so I also had 2.3.2 and 2.3.0 installed. It might well be that the issue was there in 2.3.4 already because I just updated but did not "use" the version,
If you need any other detail just let me know!

closed 2020-10-22 by stale[bot]