opentx/opentx #7332: Companion: <total duration> computation in Telemetry logs window is wrong

MarkusPfaff, 2020-01-24:
One picture says more than a thousand words:
[image]

Best Regard
Markus

3djc, 2020-01-24:
What’s wrong ? The 4:06 covers the period from 13:51 to 17:57, which is right, so not sure what is wrong there ?

MarkusPfaff, 2020-01-24:
Yes, this is about the wording and what we want to get from the data.

A "total" is a sum: total duration is the total of all durations thus the sum of all durations:
05:14 + 1:14:10 + 00:34 = 1:19:58
i.e. more or less the "flight time" you had this day (when taken in an optimistic way).

The time span the sessions are spread over is the time from the start of the first session to the end of the last session including all pauses in between. But this is not the total of the durations of all the sessions.
from 13:51:49 to 17:57:20+00:34 = 4:06:05

This is the time span the transmitter was first switched on until it was finally switched off. In many cases this will include a test at home before you put everything in the car to drive to the flying site and (like in the example above) a short test at home, because something needed adjustment. This may be more interesting for your family than you yourself, because they have proof that you are wasting too much time on your hobby.

PR #7355 (2020-02-06): Fix #7332
Change Total time to Time span when displaying logs

Fixes #7332
commit 00cd0f91 (2020-02-06): Fix #7332 (#7355)
closed 2020-02-06 by 3djc