opentx/opentx #7333: Jumper T16 MDL button on OpenTX 2.3.5

FPVxfun, 2020-01-25:
Dear all, on the last OpenTX version, MDL button does not allow to enter Model Menu settings by short press. It only allows to enter display layout configuration. I hope you can fix It soon because in this condition the T16 could not be used. Thanks in advance

3djc, 2020-01-25:
You have likely a flat cable issue. Please go to RADIO/HARDWARE/Keys (at the bottom) and watch what happens when you push MDL

closed 2020-01-25 by 3djc

FPVxfun, 2020-01-25:
Dear friend, all was working well up to 2.3.4 version. MDL Key Is correctly working with long press

kilrah, 2020-01-25:
Entering model setup menu has always required a long press by design.

FPVxfun, 2020-01-25:
Dear kilrah, you foto the reason! T16 gas BIG problema of soldering residual on PCB. When I got It for the First Time the scrolling was not working, now a push button. The problem Is the same. In order to solve all of them, I had to unmount/clean/reinstall ALL the PCB on the radio!!!! Honestly I think that Jumper has big qualità problems on its best-seller T16!!!!