opentx/opentx #7339: T16 Enter button acting as return button

Itsyaboijake, 2020-01-27:
Hey all, 
I've been experiencing the problem with my Jumper T16 pro that my enter button of my school wheel is acting exactly the same as my return button in all situations. I'm unable to update firmware due to returning out of the bootloader menu when trying to write firmware. I believe my USB port is also bad as I'm hearing as though the electrical connection is engaging and disengaging when connected to the computer or when I wiggle it. 

Summary: I can't use my radio at all as there is no enter button and I cant reset the firmware due to lose USB port and enter key working as return key. 

Please help 🙃

**General information**
 - OpenTX version (e.g. 2.3.5)
 - Hardware (Jumper T16)

3djc, 2020-01-27:
Not an OpenTX issue, but likely, one of your flat cables has moved. Try removing and reinserting the ones going to/from the two side switch/scroll area.

closed 2020-01-27 by 3djc

awshima, 2020-03-14:
Same thing here, the enter key started to work as a return key.
It was working fine, I wrote the models from Companion to the radio then I noticed this behavior.
Something different that I did was to copy one model from my Q X7 to my T16, I had both radio's model list opened at Companion, and I did a Copy Model/Paste Model from one list to another.

3djc, 2020-03-15:
T16 buttons issues have only one source : the ribbons connectors

awshima, 2020-03-15:
Opening mine and reattaching the ribbon cable (which looked ok, but...) did the job.

bhulter, 2020-03-17:
Thanks guys. Same problem.... Same solution. Ribbon looked fine but cycle the connection and the problem went away.

jimconner, 2020-06-08:
+1 Same issue, same fix. Thanks for sharing the solution.

Vitaly591, 2020-11-13:
Аналогичная проблема, аналогичное исправление. Большое спасибо, что поделились решением!

Gruefus, 2021-02-22:
I really appreciate I found this thread. The proposed fix was easy and did the trick. Thank you.