opentx/opentx #7340: Support of Devo Telemetry

richardclli, 2020-01-27:
I checked the documents and settings, but cannot found any supports about the telemetry function of the Devo protocol.  It is not supported yet?  The version that I have tested is 2.3.5.

richardclli, 2020-01-27:
I can see if I can contribute if this is not yet supported yet.

3djc, 2020-01-27:
Isn't that more a Multi request ?

richardclli, 2020-01-31:
Just figured out that this is the multi module's function, and checked that it is not supported yet.  I will follow up this issue and try to develop a fix for this in the multi module project.

richardclli, 2020-02-18:
Seems this is not only a multi request.  The telemetry decoding logic is in OpenTX source code and not only in the Multi source code.  I think I can solve this anyway ^.^

3djc, 2020-02-18:
yes, but those are usualy added to OpenTX by multi team, as it requires detailled knowledge of those protocol that we usualy do not have

richardclli, 2020-03-10:
One thing I think it is not good is the telemetry receiving code and the telemetry decoding code splits into both the open-tx and the multi-protocol module code.  I think it will be better to have a unified telemetry protocol that bridges the module and opentx without the need to split the coding into 2 parts for different protocols implemented.

richardclli, 2020-04-10:
I have added telemetry support for Devo in multi module using Frsky Hub, therefore only battery voltage is supported, mainly for microheli voltage feedback.

closed 2021-01-06 by stale[bot]