opentx/opentx #7341: Proposal: make telemetry lost detection time adjustable

Allerhopp, 2020-01-27:
- SBus lost frame bit is no more true with V2 FrSky firmware from may measurements
- FrSky 0.9s lockouts could have been detected much earlier with a shorter detection time

Is there a chance, to make the detection time shorter, 0.5s or 0.7s? Or adjustable for advanced users in a more hidden place? Additionally a LQ index based on returned telemetry frames percentage would be great, similar like it was introduced for MPM. But based on frames with a gliding average. It has to consider the different frame rates, imo 111/s and 140/s. From my pov the telemetry link quality is a 1:1 copy of the control link quality if hardware has no issues.

3djc, 2020-01-27:
If I understand correctly, nothing can be done on OpenTX side, all we do is convey values comming from the module

Allerhopp, 2020-01-27:
OK, eventually I'm wrong here. I assumed that if a control frame fails, the receiver does not send a telemetry frame. If the receiver sends a telemetry frame anyway, my suggestion is bs of course. 
My other assumption: Telemetry lost is triggered after a certain time (=number of lost telemetry frames (about 1s)) presumably does also not work in this case.

3djc, 2020-01-27:
No relation between lost frames and telemetry, I'm afraid. You need to take this one with FrSky I'm  afraid

closed 2020-01-27 by 3djc

kilrah, 2020-01-27:
Actually with FrSky stuff if there is no control frame received no telemetry is sent.

Allerhopp, 2020-01-27:
True, it is not evidence, but it is an indication and it is better than nothing imo.

kilrah, 2020-01-27:
We also don't know for sure what the module is doing, it could filter stuff, duplicate it... Wouldn't expect it to but would need to be investigated.

I mean if you want to look at that kind of stuff it's easy enough to modify the OpenTX code and build test firmwares yourself.

Allerhopp, 2020-01-27:
I can do the investigation, but I need to find someone, who is smart enough to do the code changes and compiling for me, this is not my world. In case of success, I will chime in again.