opentx/opentx #7343: Jumper T16 does not recognize internal module after 2.3.4 upgrade.

jjzorro, 2020-01-27:
The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

3djc, 2020-01-27:
Download again and reflash WITH internal multi option ...

closed 2020-01-27 by 3djc

jjzorro, 2020-01-28:
Interesting. Tried to reflash but get a warning that file cannot be
downloaded for “unknown reasons".