opentx/opentx #7345: T16: missing channels in DSM protocol when binding with option "Disable Ch. map"

sperlinh, 2020-01-28:
**Describe the bug**
Jumper T16S with internal MPM
When binding with option "Disable Ch. map" switched 'on', then e. g. THR channel can not be controlled.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "model setup"
2. Internal RF: Mode: MULTI, DSM, X 11ms or 2 22ms
3. Used Module Status: V1.3.0.67 TAER (but should be 53)
4. Select option "Disable Ch. map"
5. Click on Bind button
6a. For DSM2 (6 channel DSM2 lemon RX): Result: THR (Ch. 1) cannot be controlled
6b. For DSMX (7 channel Stabi DSMX lemon RX): Result: THR (Ch. 1) and GAIN (Ch. 8) cannot be controlled

**Expected behavior**
All channels shall be controlable (as shown in the servo monitor of the transmitter)

**Work around**
Step 4
De-select the option "Disanble Ch. map"
Step 5
Bind the RX
Step 6
All channels can be controlled (as expected)

**General information**
 - OpenTX version (2.3.5)
 - Hardware (Jumper T16S)
 - Multi Protocol version V1.3.0.53 (but displayed as V1.3.0.67)

3djc, 2020-01-28:
It looks more like a multi possible issue. @pascallanger any hints ?

kilrah, 2020-01-28:
> Multi Protocol version V1.3.0.53 (but displayed as V1.3.0.67)

The version displayed is the version sent by the module, so if you see 67 your module isn't on 53.

pascallanger, 2020-01-28:
Hummm, this is strange.
1. You don't need to redo the bind since "Disable ch map" is a dynamic setting. So you can uncheck/check it and observe right away the baehavior.
2. I see that you are already in TAER so checking/unchecking "Disable ch map" has basically no effect at all... Nothing is changing.
3. "Disable ch map" only affects CH1 to CH4 and has no effect on CH8. It's also independent of the protocol and sub protocols currently in use.
4. I've tested:
 - module in AETR with mixers set to AETR: Disable ch map off -> THR radio CH3 is driving the THR(CH1) of the RX, Disable ch map on -> AIL radio CH1  is driving THR(CH1) of the RX. Which is the expected behavior.
 - module in TAER with mixers set to TAER: Disable ch map off -> THR radio CH1 is driving the THR(CH1) of the RX, Disable ch map on -> THR radio CH1  is driving THR(CH1) of the RX. Which is the expected behavior.

May be you could post a video of your behavior?

sperlinh, 2020-01-28:
Hi all,

I’ve tried to reproduce the issue, but it did not happen again. All channels are working fine regardless of the setting of the switch “Disable Ch. Map” during binding. You can close the issue.

Best regards and thanks a lot for the quick and helpful response!
Herbert

closed 2020-01-28 by 3djc