opentx/opentx #7352: Wrong numbers when speaking in Portuguese after 99

michelsneto, 2020-01-31:
Wrong numbers when speaking in Portuguese after 99
Hi friends,
The portuguese speaking numbers are wrong after 99 example:
The number 100 is spoken 200 in portuguese;
The number 200 is spoken 300 in portuguese
The number 300 is spoken 400 in portuguese
Aways increseing 100 on counts
The number 1900 sounds very funy 1000 and 1000

kilrah, 2020-02-01:
Do you have the correct sound files loaded for your OpenTX version?

michelsneto, 2020-02-01:
Hi Kilrah, thanks for help!
Yeap, it were my first try , I did a double check if the audio pack ware the right one related to the firmware version. I did not saw details in the code but I guess that it could be related to the difference in Portuguese and a bug in code. In portuguese the count 100 is spoken "cem" after that is diferent 101 is spoken "cento e um",  102 = "cento e dois", ...135 = "cento e trinta e cinco"
 etc.
In other words for exact count 100 is spoken "cem", for counts greather than 100 the word "cem" change to "cento"  plus the the trailing count. I tried just to raname the files but it work onli until the 800 i did not work completely becouse de code selects the same audio file for numbe 900 or 1000 tha´s why I said that the numbe 1900 is spoken "mil e mil" or if it ware in English spoken "one hundred and hundred".

3djc, 2020-02-01:
You'll need to provide us more guidance. How is 200 said ? 201 ?

Please explain more the 900/1000 issue

3djc, 2020-02-01:
Also, we will not fix 2.2.4, you should try 2.3 branch. I just pushed a test fix in 2.3.6 nightlies, could you try with that simu (please use corresponding sd pack)

michelsneto, 2020-02-01:
Wen these issues happened Iin 2.2.4 tried to move to 2.3.5  branch but the telemetry simulator on companion  is not working on these branch 2.3 (it seems to be a known issue wainting to a fix as far as I could read on foruns). That´s  why I did a step back again to 2.2.4 because I´m aways making a lot of simulation tests with telemtry resources. I did not test the way of speaking on 2.3 yet (I did a downgrade on firmware back to 2.2.4 on taranis) but Iwill do the teste on 2.3, at least on companion to provide you some data.

Giving more details about the issue, it seems to affect just Portuguese, it is OK in English.It never speaks the number 100 "cem" on Portuguese it aways speak next hundred, it is ok until 99, after that (100) is spoken as 200, the displayed value 200 go to loudspeaker as 300, the displayied value 201 sounds 301; displayied 500 sounds 600; displayied 501 sounds 601 aways adding 100 to spoken count. When displayied reaches 900 it souds like 1000, 901 sounds like 1001, 902 as 1002. When displayied count reach 1000 it woks Ok until 1099. when display reach 1100 it sounds like 1200;  displayied 1101 sounds 1201 and so on. Is it clear ? I really appologise for the very poor english.

michelsneto, 2020-02-01:
I just finished a preliminary test on Companion 2.3.5 with sound pack included on SD card image x9d 2.3V0025.It has the same issue speaking hundreds counts in Portugese.

3djc, 2020-02-01:
Did you compile yourself or download? Because the changes made will only be in next nightly

michelsneto, 2020-02-01:
I did not compile it by myself, I just download the last stable version windows install file  "companion-windows-2.3.5.exe".
I forgot to mention in all cases I noticed the issue using play value special function

3djc, 2020-02-01:
Please test with next nightly, 2.3.6N117

michelsneto, 2020-02-02:
Sorry, I'm not aware about this GitHub search resources and I did not find the file 2.3.6 N 117. I'm windows user and never could realy prepare and configure a computer to compile the Open TX firmware or Companion (I know,  it´s a shame   :(       ). Is it possible to inform a link for an windows installer for this 2.3.6N117 ?

3djc, 2020-02-02:
https://downloads.open-tx.org/2.3/nightlies/companion/windows/

michelsneto, 2020-02-02:
3djc,
Good news, this issue does not happen in 2.3.6N117!!!! I test it by changing gvar1 and play value special function from 0 to 1024 and seems to be OK.
Unfortunately the telemetry simulator still not working but this is another story...
In this mean time I will keep the 2.2.4 with thisb bug on speak number until 2.3 branch telemetry simulator be fixed.

Many many thanks!!!!!

michelsneto, 2020-02-02:
This issue is not fixed in 2.2 branch but is seems to be fixed there in 2.3 should I click close here in 2.2.4 ?

bsongis, 2020-02-02:
Thanks for the report, yes we only maintain 2.3

closed 2020-02-02 by bsongis