opentx/opentx #7354: Write Firmware -> Directory is missing!

jj-uk, 2020-02-02:
Taranis X9D+ (2015)

I've just updated to 3.3.5, and formatted the SDCard and loaded "sdcard-212x64-2.3V0025" contents to the SD Card.

From the bootloader (which is version 2.0.15), if i go to Write Frimware, or Restore EEPROM, i see "Directory is empty!".

I have reloaded the previous firmware 2.2.3 back to the radio, and restored the original SD card contents for this version to the SD card, and i've tried a new SD card, but nothing is working.

I flashed a receiver with the radio yesterday, but today, after the update to 3.3.5, I get "Directory is empty!" error.

Can someone suggest a solution? What causes this error? From the SD card viewer in the radio menu, i can browse the SD card files.

jj-uk, 2020-02-02:
I've managed to update the bootloader by dropping it in to the FIRMWARE folder on the USD card and then executing the bootloader update from the SD Card browser (not from the bootloader).

Once the bootloader was updated and power-cycled, I can again access the folders.

closed 2020-02-02 by jj-uk

kilrah, 2020-02-03:
Directory name changed in 2.2.

gusemma, 2020-06-03:
My X9D+ will only open Bootloader 2.0.9 with Write Firmware, Restore EEPROM, and Exit. From there I get "Directory is missing" and that's all, no other functions. What must I do?

3djc, 2020-06-03:
Update bootloader.

mraiman001, 2023-10-15:
I had this same issue, I did update bootloader, but still was getting "directory is missing"
What solved the problem for me was changing the file location on my sd card.  I initially copied the sd card contents to a separate folder on the sd card.  Apparently, it didn't like that.  I corrected this and copied the sd card contents to the card without using any subfolders and presto, it worked.  Also, as recommended by TweetFPV, I shortened the file name to 8 characters.