opentx/opentx #7357: Aileron Option for Elevons (delta wing aircraft),and flying wing missing from model wizard on Horus X10S

builda1, 2020-02-03:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
When using the model setup wizard on the radio I noticed the option to choose “delta wing”(elevons) and “flying wing” has disappeared from the wizard on my X10S. Have. Not checked taranis X9D+

Running 2.3.5

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '.create new model'
2. Click on '.plane'
3.See error-first question used to be to choose the type of wing, followed by then which aileron channels to use

**Expected behavior**
I expected the above mentioned steps-ended up using companion in this case to create model

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.3.5
 - Hardware X10S (have not checked Frsky X9D+ But I should)
 - OpenTX Companion 2.3.5 on windows 10

3djc, 2020-02-03:
We are counting on the community to expend features using lua. Wizards are definitely part of that. So all it takes is someone volunteering for the job and make one

closed 2020-02-03 by 3djc

builda1, 2020-02-03:
It USED to be there until I updated my firmware with this so-called critical update so I don’t know what you guys have changed. Since I’m not a software dev I can’t fix it myself hence I reported it here-I guess that is a waste of time! Just Hope nobody else tries to set up a delta wing plane with elevons and rudder! Oh well, I tried to help🤷‍♂️

3djc, 2020-02-03:
Then you probably had a sd pack with one on it. What where you running before?

kilrah, 2020-02-03:
You must be confusing with another radio, AFAIK that was never an available setting in the one for the X10/X12.

3djc, 2020-02-03:
Unless maybe someone added some in a pack we don’t know about, hence my question on what he was running

mortehoe, 2020-02-03:
[image]
Mine shows in companion 2.3.5 but not on the radio if this helps

3djc, 2020-02-03:
Yes one is available in companion, for all radio, the issue here is on the radio itself

builda1, 2020-02-03:
Ok, so thanks guys for your comments, it appears it has always been missing on the radio on the X10S (I only got mine a short time before xmas). I still have my X9D+ I will look on there to make sure it’s still an option on there as it was when I last set up elevons from memory. Thanks guys for clarifying it’s always been missing from the wizard on the radio-hopefully someone in the community can rectify it someday-sorry, I don’t know lua so I can’t help. Cheers