opentx/opentx #7362: Crossfire tools crashed when using with QX7 and TBS micro TX / open tx 2.3.5

hervememain, 2020-02-05:
I have several error message when I try to change the settings for the crossfire (micro tx nano rx)

go to Tools / Crossfire Config

two lines appear : XF Micro TX and XF nano RX

when i select nano rx the screen become completely white, and if i press the scroll wheel I have the message : "Script Killed CPU limit". I can't modify the settings of the nano RX

when i select XF Micro TX it is the same : blank screen and the same message "Script Killed CPU limit" when i turn the scroll wheel

if on the blank screen, rather than using the scroll wheel I press the enter button I have an error message : 

Syntax error script
?:0 
attempt to index a nil val...

I don't see the end of the message because the line passes on the side of the screen

The binding between the RX and the TX seems to be good because the led on the nano rx goes from red to green when i turn on the RC

Moreover, when I go betaflight the RC is not recognized

**General information**
 - OpenTX version 2.3.5
 - Hardware : taranis QX7
 - OpenTX Companion 2.3.5
- SD card files : 2.3V0025

hervememain, 2020-02-05:
[image]
[image]

3djc, 2020-02-06:
Have you set your interface to 115 ? (Screen hardware, Max bauds to 115k)

hervememain, 2020-02-10:
Thank you, fixing the baud rate to 115k works.
I also added the inverted mod on the QX7 and it works fine at 400000 bauds

Thank you

closed 2020-02-10 by hervememain

jasc76, 2020-04-15:
What can it be if this message comes on a qx7s with hardware mod ?
TBS Crsfshot OpenTX Build.

3djc, 2020-04-15:
It means the communication is not working as it should, so hard to say