opentx/opentx #7363: Where are my sensors?

Buswab, 2020-02-06:
I experienced a problem earlier today and wonder if anyone else has suffered it. I received a new X8R rec. and lipo sensor on Saturday so sat down last night to set up a new model using Open TX Companion.
Updated the X9D+ to 2.3.5 as recommended then set up the new plane. All good - no problems. Out in the workshop this am to complete the installation. All was going well until it came to setting up the sensor. The Tx couldn't find it. Thinking I'd got a bad sensor (was displaying all cells on its own OLED) I pulled the sensor from another plane. Still no joy.
OK. Try a different receiver (from the same donor plane) and, after binding, had the same problem. Re-bound it to the original plane and it couldn't be found then either. I then realised that the Tx was not picking up the RSSI or Rx voltage either.
Set up a new model for the donor doing all the programming on the Tx - all worked fine. Set up a further new model for the latest plane and Rx and the sensors were all found.
I would appear that the upgrade to 2.3.5 screwed up the original programming and the new model. Any thoughts on this.

kilrah, 2020-02-06:
You don't say either what radio or what version you updated from. Did you run a sensor discovery?

Buswab, 2020-02-06:
Hi

Radio is a FrSky Taranis X9D+. I think I updated from version 2.3.2

Many thanks

Brian

closed 2020-11-03 by stale[bot]