opentx/opentx #7369: Jumper T16 FlySky telemetry not working (FSky 2A)

Johnnysod, 2020-02-09:
Hi
I'm new to this and I'm using a EChine Trashcan FlySky version with a Jumper T16 running OpenTx-t16-2.3.5 (db7d2413). 2020-01-15

When I was using JumperTx  I had to bind using a AFHDS2A setting (from memory) and I was able to see telemetry like battery voltage, Under  OpenTx I now bind using FSky 2A but I can't discover any sensors in telemetry.
Any Ideas on what I'm doing wrong, or do you think the setting for FSky 2A has an bug compared to the JumperTx version?

John

3djc, 2020-02-09:
No known issues, on the contrary a lot of improvement have been made. Are you running a recent multi module version ?

Johnnysod, 2020-02-09:
Err I've not updated the module. I'll give that a go and see if it helps. I'll google what to do!

Incidentally I went to Opentx because my Crossfire module for my Cinecan didn't work, but it turned out the pins hadn't  lined up. I has to end then slightly to get it to connect. 
Anyway I think OpenTx is the future so I'm sticking with it

Johnnysod, 2020-02-09:
Good call. I updated the module. 
Thank you I have telemetry again.
Although, a strange [ problem occurred, I am using RETA so I downloaded that multi firmware version, but it screwed up my channel order. The AETR version works fine even though I'm still on RETA?

3djc, 2020-02-09:
Closing this Check your model channel order, not the radio one

closed 2020-02-09 by 3djc

Johnnysod, 2020-02-09:
Thanks again.
The channel order is RETA on everything but the firmware for the module. Strange but it all works!

kilrah, 2020-02-10:
It's becasue the mosule is meant to be usable on radios where you can't select order, and so that you can always use the same order in your setups but the module reorganises them according to the specific protocol used.