opentx/opentx #7370: confuse about Mixer

ZTreeRoad, 2020-02-09:
I AM NOT SURE it is a bug

1.Inputs is empty

2.In page "Mixer"
CH1->RUD Weight(100%) Switch(RUD)

In "Radio Outputs",CH1must be 100%,but it display -100%
 
**General information**
 - OpenTX version (2.3.5)
 - Hardware ( 9X Pro)
 - OpenTX Companion (2.3.5)

ZTreeRoad, 2020-02-09:
1.Inputs is empty

2.In page "Mixer"
CH1->RUD Weight(100%) Switch(RUD)
CH2->ELE Weight(100%) Switch(ELE)

3.In "Radio Outputs"
RUD down & ELE up        CH1 display -100%,CH2 display 0%,
RUD down & ELE down   CH1 display 100%
ELE can change RUD?

and i found that some switchs distube each other,such as THR/3POS/ELE,and so on

**General information**
 - OpenTX version (2.3.5)
 - Hardware ( 9X Pro)
 - OpenTX Companion (2.3.5)

kilrah, 2020-02-09:
Companion export for 9XR Pro seems broken in 2.3.5.

ZTreeRoad, 2020-02-22:
> Companion export for 9XR Pro seems broken in 2.3.5.

it is a bad news

commit cb31b10b (2020-03-27): Fixes #7370
closed 2020-03-27 by bsongis