opentx/opentx #7373: Lua Script for TBS Crossfire and Horus X10 (OpenTX 2.3.5 and earlier) - menu navigation controls (wheel) do not work.

VSCharlie, 2020-02-10:
Horus X10S Express Transmitter - OpenTX 2.3.5 (sdcard-480x272-2.3V0025.zip) and TBS Crossfire Micro.
After running the script (loading TX and RX) and running one of them, you can only move in one direction in the setup menu - to the plus (or down).
In other words, the "roller" on the right side of the horus in the script only works right (down) and does not work left (up)!
So I can only increase individual parameters on the selected line of settings - go to the right, but do not move on the line to the left - go back.
This applies to all values that can be changed by the script in TX and RT.
It's as if only the (+) button worked on Taranis Plus, and the (-) button didn't work.
All older versions of lua scripts in OpenTX behave the same way.

lichtl, 2020-02-10:
Have you tried the nightly? I've fixed the LUA scripts (maybe even just use the crossfire.lua / device.lua from the current nightly should be enough)

VSCharlie, 2020-02-10:
I tried some (see links TBS manuals), but I probably did not find the corrected.
Could you get a link to the corrected lua scripts?
Thank you..

3djc, 2020-02-10:
Take the ones from last nightly sd pack : https://downloads.open-tx.org/2.3/nightlies/sdcard/

VSCharlie, 2020-02-11:
Thank you very much.
I used "sdcard-horus-2.3V0026.zip" and it works as it should.

closed 2020-02-11 by VSCharlie