opentx/opentx #7375: Rescue Mode Horus X12S Bluetooth telemetry OTX 2.3.5

Bendau, 2020-02-10:
On openTX 2.3.5 in Horus X12S when the telemtetry of the Crossfire is redirect to the internal Bluetooth, the radio reboot in Rescue Mode.

Bendau, 2020-04-26:
Test with 2.3.6 and 2.3.7 and same problem. Last test with 2.4 TBS nightly (CRSFshot) and no problem with the rescue mode.

Bendau, 2020-05-04:
Oh Thank you ! If you need more test or what i can make to help you i'm in ;)

Bendau, 2020-05-08:
[https://www.youtube.com/watch?v=tH-KONv9hjs](url)

brainbubblersbest, 2020-05-17:
Do you use full speed crossfire with 400.000baud?
Are you trying lower that telemetry speed? 
I dont know if thats an issue when send/mirror telemetry to a bt module running at 119200baud.
Maybe some one here can enlighten me?

3djc, 2020-05-17:
the telemetry is only a portion of those 400k, mainly used for channels

PR #7652 (2020-05-31): Avoid Bluetooth fifo full (and 1ms wait in the mixer task...)
Should fix #7375
closed 2020-05-31 by bsongis