opentx/opentx #7378: Default servo travel setting

jj-uk, 2020-02-11:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
The default servo travel is +-100%. Many of my models require a different travel, e.g. +-77.8%, +-82%(1100-1900) or +- 97.7% (1000-2000).

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add a "default servo travel" setting as a per-model setting so that the servo travel is set to this value when "resetting" the servo travel value via the popup menu. Add a new menu item to the servo screen "reset all servos", which applies the default travel value, e.g. +-82%

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Manually setting required channels to +-82%.....

**Additional context**
Add any other context or screenshots about the feature request here.

loetefix, 2020-02-15:
I wish that function too.
Or an Entry in the Servo Menu with COPY LIMITS TO ALL Channels.
So you can Change all Ranges quickly.

jj-uk, 2020-02-15:
+1 for copy limits to all channels.  
So to set all, the user would set the 1st row to the required limits, then use the menu to copy those limits to all the servo outputs.

I think this would be the easiest method for the devs to implement.

loetefix, 2020-02-16:
I try to make it.Seems not to be that hard.If its ready i start a pull request.maybe they implement it.

PR #7396 (2020-02-18): Add copy min/max to outputs

[image]

This closes #7378 (please read issue for details)
closed 2020-02-18 by kilrah