opentx/opentx #7381: 2.3.5: Para Bluetooth not working for X10 express

patman13, 2020-02-12:
Still not working for me. 
X10 Ex as Master, X-Lite S as Slave —> not working
Other way round is working. 

Same as described in #7094

dev-fred, 2020-03-01:
After investigating on my bluetooth, I am sure it's not working at all, a friend loan me a X-Lite pro.
When I select Bluetooth = Telemetry in Hardware page -> PIN Code  000000 and Local addr = 806FB01B6FC6
If I do the same choice in my X10 express -> PIN Code  000000 and Local add = ---

afdofdez, 2020-03-30:
After investigating on my bluetooth, I am sure it's not working at all, a friend loan me a X-Lite pro.
When I select Bluetooth = Telemetry in Hardware page -> PIN Code 000000 and Local addr = 806FB0FE883A
If I do the same choice in my X10 express -> PIN Code 000000 and Local add = ---

in the bluetooth the my Samsung S8 it appears as "Hello" if I try to link it tells me that it has not been able to link and it shows me the remote address I suppose of my samsung S8.
[image]

dev-fred, 2020-09-27:
After changing my bluetooth module, it works like a charm, my problem was purely hardware

closed 2021-09-04 by stale[bot]