opentx/opentx #7383: losing telemetry, never coming back

guillaumevaret, 2020-02-13:
Hi, 
I have, I believe, a new issue with the Horus X10 Accst / 2.3.5 opentx-x10-lua-noheli-fr /  2.3.6N110 / X8R with the new version of frsky X8R_ACCST_V2.0.1

When I lose the telemetry (except RSSI, BtRx) when the drone is far or very close or when there is interference on 2.4Ghz, it never come back when the drone is at a normal distance, like it use to do with almost the same configuration on X9d 2.1. 
Reset telemetry doesn't work. 
Reset the flight either.
I have to restart the X10 and it will come back again. 
Once in 10 times, I had to restart the X8R too. 

Is there a way to restart the reception of the telemetry without stopping the flight ?

Thanks for your wonderful work !

3djc, 2020-02-13:
Does it work with older ixjt/x8r firmware ?

guillaumevaret, 2020-02-13:
Sorry I can't try anymore this configuration. The X8R SmartPort is under epoxy, and all the sensors to. But your right, I suspect this could be the problem. 
I never notice this before I change both at the same time the firmware version of the X10 and the firmware version of the X8R. But I can be wrong : I'm quite new with the X10.  This can also coming from a saturation of the smart port, because I have sometimes impossible values on different sensors for a 1 second duration. Like temperature 256116, or current 98122 A. I solve it adding test like L01 : TELE1: curr < 160A

3djc, 2020-02-13:
Well, if thats the case, then there is not much we can do from OpenTX, all this is managed by FrSky firmwares I'm afraid

kilrah, 2020-02-14:
There are known issues including some with telemetry loss on the 2.0.x ACCST firmwares. You'll have to wait for updates from FrSky. Several people recommend not to update to 2.0.x ACCSTjust yet.

closed 2020-02-14 by kilrah

guillaumevaret, 2020-02-17:
I did some test with X9D+ / X8R_ACCST_V2.0.1 / RPM temperature sensor / MLVSS lipo voltage sensor

So all the components are different from the previous test, to remove the hardware factor.  I was able to redo several times the " impossible values on different sensors for a 1 second duration", but it's hard to know what's the factor. 
- Could be when the power supply can't deliver enough energy. (Seems easier to do when the battery is very low 2x3.6v), but can be done with a charge battery also. 
- Could be the quality of the signal (too strong or too weak)
I can't redo yet the "telemetry loss that never come back"  in this configuration. 

Next I try to flash the firmware without removing the sensors. Obviously it doesn't work.
- "Module refuse data"  / break the firmware from the sensors.
so I did find the MLVSS.frk to put back in lapin fou SDpack
Can't find anywhere the firmware for the RPM.  RPM.frk ? 

Next I switch the firmware from the X8R to 20170210. I can't redo the " impossible values on different sensors for a 1 second duration". So one part of the problem come clearly from the firmware 2.0.1 like 3djc and kilrah said right away. 

I also maybe find a way in Opentx to restart the telemetry without stopping the flight, switching with a second identical model. I will try this next time. 

Thanks for your help and advice.

bsongis, 2020-02-17:
"I also maybe find a way in Opentx to restart the telemetry without stopping the flight, switching with a second identical model. I will try this next time."

This is interesting, please confirm!

guillaumevaret, 2020-03-06:
Not tested yet the solution, (I will) but find this post with the same problem and a very different configuration (G-RX8 + ACCESS 1.1.3). They believe it comes from the tension on the smart-port and not from the ACCST fimware (they use ACCESS fimware). So still not an opentx issue, but solve by restarting opentx.
https://www.rcgroups.com/forums/showthread.php?3468983-ACCESS-telemetry