opentx/opentx #7384: condition in logical switch not respected in GV

guillaumevaret, 2020-02-13:
Hi, 
Horus X10 Accst / 2.3.5 opentx-x10-lua-noheli-fr / 2.3.6N110

I have a condition in logical switch that doesn't work when it came from a global variable (here L12 : GV7 = 70), but work with a direct value (L13 x= 35). Doesn't work on companion either on the X10.

[image]

3djc, 2020-02-13:
The value in the current sensor is in fact 700, it is displayed as 70.0 because your sensor is set with 0.0 precision. So in you case, set your GV to 700

closed 2020-02-13 by 3djc

guillaumevaret, 2020-02-13:
Thanks a lot, it works better than my "fix". 
I was using a momentary fix adding a mix line, switching the GV and the CH.
ch32 : max ratio(+vg7)