opentx/opentx #7385: External & Internal module with CRSF protocol

jace26, 2020-02-13:
Version: 2.3.5 stable
Equipment: FrSky Horus s10x Express, R-XSR, TBS Crossfire Micro Tx, TBS Crossfire 8ch Diversity RX

I want to have dual band outputs with the internal FrSky 2.4 RF module and the External crossfire Micro Tx module. Currently, I can do this, but only if the external module is set to PPM, which means I cannot configure the module or the connected RX through the Lua script. 

I can in fact turn on the external module in CRSF with the internal module on, but there is no output on the RX until you switch to PPM or turn off the internal module.

Is there some hardware constraint that prohibits the desired behavior?

kilrah, 2020-02-14:
Yes. CRSF communicates on the SPORT line, which is common with the internal module. When the internal module is on it sends data on that line and corrupts the CRSF stream.

closed 2020-02-14 by kilrah

jace26, 2020-02-14:
They cut the line to the internal module which allows the external module to talk on it? Sounds great, Where can I find that line? google didn't find anything, but I'm open to my fu being weak.

jace26, 2020-02-14:
nm, found it.
https://www.rcgroups.com/forums/showthread.php?2922046-FrSky-Horus-X10-and-X10S-and-New-Express/page872#post42904121

may through a switch in there so when frsky updates it to require the s.port I can fix easily