opentx/opentx #7388: SBUS generation corrupted when different refresh rate selected

krzysztofkuczek, 2020-02-15:
There is a problem with SBUS signal generated by Taranis x9D+ f.v 2.3.5. Channing refresh rate cause problems with generated signal. Applying values abowe aprox. 36ms cause that interval bw frames are set below 10ms. Except of that idle line is changed from high to low...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to model setup menu
2. Click on sbus refresh rate for external module,  change to 38ms
4. observe sbus signal
5. change to different values like 20ms, 40ms, 36ms, go to point 4.

**Expected behavior**
- frequency of generated sbus frame should be adjusted according user's settings
- sbus line idle state should be low (according to inverted uart standard)

**Screenshots**
see the screenshots' all taken after signal inversion.
13ms
[image]

36ms (please note idle line state)
[image]

38ms
[image]

40ms
[image]

**General information**
 - OpenTX version 2.3.5
 - Hardware Frsky X9D+

kilrah, 2020-02-16:
Polarity is selectable.
It makes no sense to go as high as 36ms refresh, so if anything a limit needs to be set.

3djc, 2020-02-16:
20 ms max ?

krzysztofkuczek, 2020-02-16:
> 20 ms max ?

No. Plese keep 40ms. The problem with polarity is that during line idle state is changing every 2nd frame. See 2nd screen shot. 
PS. 
It would be greate to have also possibility to choose serial port speed to change it from default "standard" 100000bps 😁

3djc, 2020-02-16:
Why ? There is little point in having insanely slow  refresh rate

krzysztofkuczek, 2020-02-16:
> 
> 
> Why ? There is little point in having insanely slow refresh rate

For LRS solutions...., there is enough to have 25Hz refresh rate, and SBUS frame can be easy use to "clock" air frame sending. Just DMA transfer ending after every SBUS frame can trigger sending air frame...
When user have possibility to adjust this timing, it's limits delays... 
Regarding serial port speed, it also would be useful, cause 100 000pbs are not supported by bt modules, etc....
Kris

closed 2020-11-13 by stale[bot]