opentx/opentx #7389: Low power mode (bench mode)

jj-uk, 2020-02-15:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
When setting up a model on the bench, or when playing the simulator, my TX is transmitting at full power. This is not required; it's just waisting my TX battery.

**Describe the solution you'd like**
Add a BENCH MODE, that reduces the transmitter output power so that the battery lasts longer.
Add a warning to the screen "BENCH MODE DO NOT FLY".

**Describe alternatives you've considered**
Placing the tranmitter in RANGE TEST mode works ok for simulator flying, as this reduces the output power, but even with the volume all the way down, i can still hear the anoying beepbeepbeepbeep!
This doesn't work for setting up a model on the bench as other settings in the radio cannot be accessed when in range mode.

**Additional context**
Add any other context or screenshots about the feature request here.

3djc, 2020-02-15:
Make a model with both modules off for simu ... and that would apply only to xjt, which impact on battery life is fairly minimal

jj-uk, 2020-02-15:
If i turn off the modules, I cannot connect to my wireless dongle.

kilrah, 2020-02-16:
Battery life change would be insignificant.

jj-uk, 2020-02-16:
My taranis is using:
83mA in normal operation. 
52mA in range check display on. 
The display uses 10mA.

This is from a current clamp that may not be accurate. 
So a reduction of about 40% in range check. That's not insignificant.

jj-uk, 2020-08-14:
Stale bot bump. I'd still like this feature. It will save the battery as i fly alot of simulator time.

jj-uk, 2021-02-14:
I still believe this would be a worthwhile improvement. Charging batteries all the time is no fun. Reducing the need to charge by 40% would be a great time saver.

Combinacijus, 2021-09-06:
@jj-uk that's what [low power mode](https://www.multi-module.org/using-the-module/protocol-options#low-power-mode) does isn't it? 

It can be turned on Model Setup -> Internl RF -> Low power mode (below receiver Bind)

jj-uk, 2021-09-06:
Yes, but then the TX chirps every 5 seconds, which cannot be silenced, even when the volume is set to minimum.

Combinacijus, 2021-09-06:
Interesting my TX makes no noise in low power mode even with sounds on

jj-uk, 2021-09-06:
Just checked. I don't have low pwr mode. V2.3.11
Maybe it has been added?

I'll update later tonight.

Combinacijus, 2021-09-06:
I'm using TX16S so maybe it works only on Multimodule but I'm not sure

jj-uk, 2021-09-06:
> Low power mode is for multi module only if we are talking internal only

OK. In that case, this issue should remain.