opentx/opentx #7399: HORUS X10S Express 2.3.5 Defective antenna warning when EXT is selected using D16 ACCST

Hennyvt, 2020-02-18:
I have just received my Horus X10S Express radio now flashed with (Opentx v2.3.5) The ISRM-S-X10E version is 1.1.0/1.1.3FCC in ACCST D16 mode and with the external antenna selected I get an error message saying TX antenna problem/defective. It may take a 20 to 60 seconds or so before the radio brings up this error. I have replaced the antenna with a new one but the error still occurs.  The error goes away with the internal antenna selected. Also in ACCESS RF mode the errors are not displayed. I have reflashed my radio with Opentx 2.3.3 RC1 and then receive no error warnings in D16 mode.

In the model setup menu select D16 on the internal RF module and select the external antenna wait for the antenna defective warning. 

I am a mode-2 AETR using the FCC protocols.

[image]

kilrah, 2020-02-19:
Must be an issue with the radio and the antenna connection is actually faulty..

Hennyvt, 2020-02-19:
Hi Andre

 

I thought the same however I changed the antenna with a new one and have the same results. Using ACCESS protocols all is fine.  Also if I re-flash the radio with an older version of Opentx 2.3.3 the fault is not evident.

 

What do you think

 

Henny

DeeBee878, 2020-03-01:
I have much the same issue issue on X10s with ISRM upgrade board.  It is running Opentx 2.3.3. If I select external antenna it works well with good range.  If I restart the transmitter within a few seconds the defective anrenna warning appears. If I reselect internal antenna it works fine. Have changed antenna but it makes no difference. Unfortunately this board has the annoying authentication issue and going to 2.3.5 seems not to work.

Hennyvt, 2020-03-02:
With regards to this faulty antenna warning. I understood that the RAS reporting had been disabled in the X10s series of transmitters and is in the process of being reintroduced. Could this be causing the faulty antenna warning.
Henny

3djc, 2020-03-02:
@DeeBee878 do NOT use 2.3.3. We are not aware of any issue with 2.3.5 and the ISRM upgrade board. If you think there is one, please open an issue with as much details as possible

DeeBee878, 2020-03-15:
I have updated my X10s to Open Tx 2.3.6 and the the ISRM module with ISRM V1.1.3 form the FrSky website.  This appears to have been successful in so far as the Tx will bind to ACCESS rxs and appears to function well, although I have not tested it in the field as yet.   The problem witrh the exteranal antenna warning seems to have been fixed.  I can select either internal or external antennas and the Tx will start up and function neormally.  Good RSSI with external antena being slightly better in range testing.
Although I could intially to ACCST V1 when I first installed the ISRM it will not bind to wither the original or V2 ACCST.  I have been advised that FrSky needs to update the ISRM firmware, but I am not sure about the issue, but for now I can use ACCESS only, seriously limiting my use of the TX.
Dennis

Hennyvt, 2020-03-15:
Hi Dennis.

Are you the same Dennis who had posts on Aloft Hobbies and was attempting to upgrade his X10S with the ISRM module. Did you use a dailies version of  Opentx 2.3.6. I have been following the progress but presently there are 5 outstanding issues left to be integrated into 2.3.6. The RAS issue I believe was fixed so that may account why the antenna warning no longer occurs.

Henny

DeeBee878, 2020-03-15:
Hi Henny
No I am not the same Dennis.
I used Companion to find and download the latest.  That was 2.3.6. I then installed it to see how it went. Do you kow if the 5 issues prevent it from being used or at least test flown in fixed wing?
Thanks 
Dennis

Hennyvt, 2020-03-15:
Hi Dennis you can find the details here https://github.com/opentx/opentx/milestone/53. Where are you located. I am in Melbourne Australia

Hennyvt, 2020-03-15:
Hi Again Dennis
Just flashed my X10S Express radio with the nightly version of 2.3.6, unfortunately it has not altered my Antenna warning situation when in D16 protocol. It appears to operate fine in ACCESS protocol mode so I still have some sort of RF issue. I have ordered another antenna as well as the antenna short cable and SMA connector so will see what happens after that. I have not had any luck yet via FRsky service as my issue maybe the RF board.
Henny

kilrah, 2020-03-15:
Latest release is 2.3.5, the server had a problem for a few hours where it was serving nightlies instead of release. So if you have 2.3.6 you are on a nightly build. You should either redownload (you'll get 2.3.5) or explicitly choose to use nightlies if that fixes something for you.

DeeBee878, 2020-03-15:
Hi Henny
I am in Adelaide, Australia.
Thanks for the link.
I just checked to see if I get an antenna warning in an ACCST model setup.  I am not, but I have had a couple of messages pop up saying a 'module upgrade is required'.  
Unfortunately, I too, have not had any luck with emails to FrSky, they just seem to ignore them.  It would be nice if they would at least let us know what their intentions are and timetable for any fixes, even if that is a bit down the track.
I am a Horus RC customer and they help where they can and seem to have some channel of communication with FrSky.
I also tried another antenna, but that made no difference.  I suspect, they need to fix firmware

Hennyvt, 2020-03-15:
Go to this web site. https://www.rcgroups.com/forums/showthread.php?3507071-Important-Firmware-Update-%C2%96-ACCST-D16-2-0-0-ACCESS-also-affected/page103
There is a chap who used to repair FRsky gear. Presently his sign on is mpjf01 near the bottom of the post. Maybe you can contact him for help.

Henny

DeeBee878, 2020-03-16:
Hi Henny

 

Apparently I got 2.3.6 as a download on companion when there was a server fault.

 

I now have 2.3.5 installed following advice from Andre Bernet.

Also corrected my error in installing ISRM 1.1.3 to 1.1.5.

Mt radio now seems to work OK with ACCESS and ACCST V1 receivers.

There is still a problem with ACCST in D16 mode but apparently FrSky is working on that with German users doing beta testing.  Hopefully it will be available in the not too distant future. 

My Taranis Plus on 2.3.5 is having a problem with ‘sensor lost’ (voltage telemetry) and RTC battery voltage warning.  These are apparently known issues and I am still investigating.

 

Dennis

DeeBee878, 2020-03-16:
Thanks Andre.

I have now installed 2.3.5 and it seems to be working as expected.

Dennis B

Hennyvt, 2020-03-16:
Hi Dennis
Great you now have Opentx 2.3.5. version 2.3.6 is coming and fixes several more X10S issues but not show stoppers presently. The ISRM issues are annoying but not terminal. You can roll back your ISRM firmware if you want to get all your sensors working again until the new ISRM firmware is released. I now have the new Express X10S which also has these issues. However I still have an additional issue with my radio where I have a defective antenna warning in D16 mode. Not to confuse this with the warning asking if you have your external antenna connected. 
Henny

kilrah, 2020-03-16:
>  You can roll back your ISRM firmware if you want to get all your sensors working again until the new ISRM firmware is released.

As mentioned multiple times DO NOT use an Upgrade ISRM on any firmware older than 1.1.5, and do not recommend people do that.

vasyly71, 2020-06-30:
Hi, i got same problem today -
   Internal module: ISRM-S-X10E, 1 again .1.0/2.1.0 FCC
   Opentx-x10express-2.3.7

Was constantly getting TX antenna problem in D16 mode. When switched to ACCESS no wornings. 
After i reflashed 2.3.7 again warning was gone.

Hennyvt, 2020-06-30:
Since upgrading my X10S to ISRM ex the FRsky web site to v2.1.0 and Opentx version 2.3.7 I have not experienced any issues. I have been using ACCESS and RX6R with no issues. When using ACCESS I have reset the RSSI warning limits to 40 and 37 db.

vasyly71, 2020-07-01:
Looks like it was a legit warning. Soon after i re-flashed and then flashed latest version OTX 2.3.9 the worning returned. 
When i put antenna from R9M module message did not appear. When i put antenna from taranis x9d-2019-se message came appear again. 
I slightly bent pin in the RF connector on transmitter side screw on original antenna back and warning was gone.  
I would love to try ACCESS but my receivers are S6R and RX8R-pro and they do not have ACCESS firmware upgrade, at least yet.

 I will be ordering another antenna, just in case.

Hennyvt, 2020-07-01:
Be careful swapping antennas the R9M antenna is for the 900meg transmission and should not be screwed onto a 2.4gig transmitter . Bent pins is never good.

closed 2021-06-04 by stale[bot]