opentx/opentx #7400: channel output reverse not working on radio xlite pro

mortehoe, 2020-02-19:
Reversing a channel output (channel 9) using S1 slider on my xlite pro does not change the output direction.
It works okay in companion when I load the model from the transmitter and reverse the output there.
I am using opentx 2.3.5 connected to a Frsky S6R and setting the gain (channel 9) on S1 with weight 50 and offset 50 and cannot change output direction on the slider.

3djc, 2020-02-19:
How exactly are you trying to invert on the radio ? Just tested it, it works fine

mortehoe, 2020-02-19:
On the outputs page invert direction.It works fine on companion but not on my radio.Attached is hopefully the otx file taken from the radio .
[pro quad.zip](https://github.com/opentx/opentx/files/4226873/pro.quad.zip)

3djc, 2020-02-19:
Well, when you increase S1, ch9 decreases

mortehoe, 2020-02-19:
I want the slider S! to work in the opposite direction.Are you testing this on an Xlite pro or just in companion.
When I change the direction in companion it works correctly by reversing,then when I change on the radio in exactly the same manner the direction does not change.

3djc, 2020-02-19:
You are doing it wrong... S1 set has you have does 0 to 100. When you invert that in outputs, it correctly invert it so does -100 to 0

What you want to do needs a negative weight in input or mixer, and no inversion on output

closed 2020-02-19 by 3djc

mortehoe, 2020-02-19:
Perfect.That solved the issue.
Thank you so much for your help.
Sorry for raising as an issue, when the operator was the issue :-)