opentx/opentx #7404: I heard from domestic players that opentx will support Chinese language, right? Thank you.

Zhuwenpeng055, 2020-02-20:
If so, when can we use the Chinese version of Opened? We like Opentx very much, thank you

3djc, 2020-02-20:
Not before 2.4 at the earliest

Zhuwenpeng055, 2020-02-20:
That means Chinese won't be included in version 2.3, right?

kilrah, 2020-02-20:
Correct

bsongis, 2020-02-21:
Yes it's there in 2.4 already. Japanese and Korean will be supported as well, the code is ready. We will need help for translating all the strings, as noone speaks Chinese (nor Korean and Japanese) in the team.

iakuf, 2020-07-23:
maybe I can help for translating, What i can do ?

iakuf, 2020-07-23:
I modify and create opentx/radio/src/translations/cn.h.txt ?

3djc, 2020-10-08:
Radiomaster is working on Chinese language translation for 2.4, maybe you can review those when they are done @iakuf ?

RadioMasterRC, 2020-10-08:
Thanks 3djc, we are working on this now and will keep everyone posted.

closed 2021-09-04 by stale[bot]