opentx/opentx #7405: Dual Telem possible?

jace26, 2020-02-21:
Okay, I don't follow all this, but I have referenced this thread:
https://github.com/opentx/opentx/issues/3752

And have done the mod to remove the S.Port line as is detailed here:
https://github.com/opentx/opentx/issues/7385

Now, I can connect with the internal RF to an r-xsr in access, and an external TBS crossfire micro TX to a TBS 8ch Diversity RX, and it all works great, except that I can only get telemetry from the external unit if it is on CRSF, and only get telemetry from the R-XSR if the external is on PPM. Is this inherently a hardware constraint, or is there a possibility of getting both at the same time? 
@kilrah maybe you have some insight?

3djc, 2020-02-21:
I do not think this can work just yet. That would require two types of telemetry to be active at same time, and while this will be possible at one point in time, I do not believe it is ready yet

jace26, 2020-02-21:
I am just wondering if it may be possible to do at present it doesn't appear to work of course.

One thing I notice in the main telemetry menu is that there is a selection for telemetry source, which tends to be default. It would be nice to have, at the very least, some sort of priority selection with a failback whereby dropping the primary link reverts telemetry to the secondary link. 

Ideally one could get a full set of the primary and secondary telemetries simultaneously. 

I notice that when going to add a custom telemetry sensor, there is an non-selectable field which announces the sensors origination (internal or external).

kilrah, 2020-02-21:
There would be no primary/fallback, the intent is to display the values from both sources independently.

jace26, 2020-02-21:
> There would be no primary/fallback, the intent is to display the values from both sources independently.

yeah, I said that is the best case. The primary/fall back has some practical implications in my use case. 

The question, again, is can it be done?

3djc, 2020-02-22:
Currently no, in the future, yes, but it will not be primary/fallback, it will be both source at the same time

closed 2020-11-20 by stale[bot]