opentx/opentx #7408: No Logging of sensor data with Jumper T16 pro Internal module. When I look at the Sensor edit screen it is showing External module and cannot be edited.

rbayer62, 2020-02-23:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

3djc, 2020-02-23:
It is indeed showing external, but that is for info only. Logging works

[image]

rbayer62, 2020-02-25:
I hadn't set up logging in a long while. I had forgotten to enable in
Special functions. Thanks for the comment that jogged my brain cell.

closed 2020-02-25 by kilrah