opentx/opentx #7410: Reduce screen consumption in bootloader mode

mric3412, 2020-02-24:
When connected to a PC in bootloader mode, color LCD screen (e.g. T16) is all white : the battery is unnecessarily consumed.
[image]

It would be great to invert colors of the 2 bootloader screens ("WriteFirmware/Exit" option and "USB connected" message) in order to display allmost dark pixels and reduce the consumption.

Alternatively, having the option of simply turning off the screen (all pixels displaying in black) would be good for saving energy

This proposal concerns radios with large color screen (Horus, T16 etc.)

3djc, 2020-02-24:
It is not oled, so black pixels do not consume less...

mric3412, 2020-02-24:
Ah! Sorry, it's not very intuitive... if it's an LCD backlight, it may even consume more when displaying black !? ...

3djc, 2020-02-24:
Consumption is extremely similar, mostly happening in backlight, not the lcd panel itself. All black screen can generate a little more heat, but that is not relevant in this case

mric3412, 2020-02-24:
OK, thanks, I learned something ! It only remains for me to close this proposal...

closed 2020-02-24 by mric3412