opentx/opentx #7413: Blue led always on on receiver

vradermecker, 2020-02-26:
Hi,

I'm using an Horus X10S Express with OpenTx 2.3.5.

All my receivers (RX4R, RX6R, G-RX8) have been flashed with Access firmware.

When registering/binding a receiver, the blue led is always on.

It should be the case only in HS mode and not in FS mode.

The 9ms checkbox in receiver options doesn't change anything to this behavior...

Is it a bug in OpenTx (always in FS mode), an error in Frsky documentation or a misunderstanding from my side... No clue but I wanted to highlight this. Indeed, if the HS mode is always on, it could be a problem when using analog servos...

Thanks in advance.

Regards

Vincent R

kilrah, 2020-02-26:
OpenTX has 0 influence on what the receiver does.

closed 2020-02-26 by kilrah

vradermecker, 2020-02-26:
Ok thanks. Even if OpenTx sent always the parameter 9ms to the receiver (ignoring the value of the checkbox)?

Thanks anyway for your feedback

kilrah, 2020-02-26:
The parameter is not stored in OpenTX, it's always requested from the receiver for display. So if you power everything off and back on and it's still how you set it then the receiver stored it correctly.