opentx/opentx #7421: 9XR PRO RTC bug

vitusaleixo, 2020-03-04:
Hello
I have updated my 9XR-PRO from 2.2.4 to the last 2.3.5 with the 2.3.5 companion. It flashes nicelly, but when i set the RTC to the actual date-month-year, and restart the radio, goes back to 1970-01-01.

kilrah, 2020-03-04:
Dead RTC batt?

vitusaleixo, 2020-03-04:
No, thought that too. Opened It up and the battery is 3v. I have downgraded yesterday top 2.2.4 and the issue dissapeared

Obter o Outlook para Android<https://aka.ms/ghei36>

AvaraKaisa, 2020-04-18:
Same problem using 2.3.7.

pfeerick, 2020-05-10:
Same here. Was working prior to the update.

pfeerick, 2020-07-10:
Still not working on 2.3.9

serbyk, 2021-03-06:
I have the same problem. ver 2.3.11

pfeerick, 2021-03-07:
I've basically given up on OpenTX with the 9XR-PRO... I'm back on 2.1.10, as I was been chasing an issue where the voice telemetry randomly cuts out after several minutes (was speaking the battery voltage and mah every 15 seconds and the minute timer), as well as no... swapped SD cards, changed FW versions several times... I wanted the 2.3.x firmware as it added the USB support, but went back to 2.2.4, and then tried 2.1.10... it's been a while since I used the 9XR-PRO, so I don't know if that was the last properly working version for it... or if I swapped to the Radiomaster TX16 and haven't used it since. Should revisit it... it is still an excellent transmitter.

pki791, 2021-08-14:
Hi. Is there a chance to look at this issue? I can confirm it.

3djc, 2021-08-14:
I do not think any of the active developers does have a 9XR-PRO, which is what explain the lack of progress

pki791, 2021-08-14:
If any of the developers have time to do it i will assist, i can test every version and check which version it stopped working, then it should be not difficult to find the change which cause it not to work.

pki791, 2021-08-31:
Any1?

rotorman, 2021-08-31:
@pki791 disclaimer - I have no experience with 9XR-PRO.
By looking at OpenTX source code, I see 9XR-PRO is built with CMake option -DPCB=9XRPRO:
https://github.com/opentx/opentx/blob/ad64ce0ced3067517a81b9272d031c27bf631196/radio/src/targets/sky9x/CMakeLists.txt#L19
When I look at the board init, then real-time-clock init is called only for AR9X:
https://github.com/opentx/opentx/blob/ad64ce0ced3067517a81b9272d031c27bf631196/radio/src/targets/sky9x/board.cpp#L393-L395
Very likely _rtcinit()_ needs to be called also for 9XR-PRO. As there are no devs for 9XR-PRO, can you try to setup the build environment yourself and give it a try? See e.g. https://github.com/opentx/opentx/wiki/Compiling-OpenTx

pfeerick, 2021-08-31:
I'm very tempted to do this, but it's IMO the most trivial of bugs in 2.3.x my 9XR-PRO - I was more concerned with the ones that made the tx lock up completely so it now runs 2.1.9 IIRC. I should have a look also though since I have the hardware.

pki791, 2021-09-18:
I will try to setup the build environment and try it.
What lock-up are You talking about? I have not expereienced any lockups.

pfeerick, 2021-09-21:
https://github.com/opentx/opentx/issues/8563 is one... https://github.com/opentx/opentx/issues/7772 is another, and https://github.com/opentx/opentx/issues/7594 is another. 

Unfortunately this repo is using the stale bot to auto-close issues, so they are closed even though they are still present, which is why the "open issues" count is artificially so low... I do appreciate that it's not practical to fix it if don't have the hardware, but I don't believe it should be automatically closed without at least documenting it as a known bug, and identifying a known safe version for anyone who still uses that perfectly functional transmitter as a transmitter. Same as it has USB joystick support now IIRC, but if that were used, you'd want to not use the transmitter for anything else since it is inherently unstable.

ArtCodeZen, 2022-01-03:
Same problem the RTC does not work if you turn off the radio and turn on again the data and hour reset to 1970, openTX 2.3.14. It's just working with ER9Sky software without problem.

jarno-r, 2022-08-14:
I didn't even know my 9XR Pro had RTC! Enabling the rtcInit() makes RTC work. This could be fixed by changing the relevant code to 
````c
#if defined(PCBAR9X) || defined(REVX)
  rtcInit();
#endif
````

I'm hacking together a version of OpenTX 2.3 that works on the 9XR Pro over here: https://github.com/jarno-r/opentx. 
I might post PRs to opentx, depending on how much time I can spend on this and whether they are getting merged.

pki791, 2022-08-15:
> I'm hacking together a version of OpenTX 2.3 that works on the 9XR Pro

Good work, can You please enable some form of communication? Buf report or feature request? Can You please share instructions how to compile it in 2022?

jarno-r, 2022-08-16:
> 
I've added build instructions to the readme and turned on Issues & Discussion. The default branch (9xrpro-fixes) should now be good to use and contains the fixes mentioned in the readme.

pfeerick, 2022-08-16:
Just as a FYI in case you've not encountered it yet... [vitas/opentx-docker-build](https://github.com/vitas/opentx-docker-build) is a dockerised alternative. And further that, I would humbly suggest looking at [pfeerick/opentx-fw-build](https://github.com/pfeerick/opentx-fw-build), which is a fork of the former project which adds pre-built docker images served via github.