opentx/opentx #7422: sporadic message: Module upgrade required

ParkerEde, 2020-03-04:
Hello everyone,
I converted my Horus X10S to the ISRM module, installed OpenTx 2.3.5 with the options internalaccess, EU and LUA and then flasheded the ISRM firmware update 1.1.5. Everything works as far as it should. I have already bound some D16 receivers but also three Access receivers, all without problems.
I occasionally get a message window: Module upgrade repuired
While this message is on the screen, the model can continue to be controlled.
An external module is not available and is also not activated in the model memory.
The message also comes with different models.
Does any of you have any idea where it comes from or whether I have to worry?

[image]

If I then press RTN, everything continues as normal.

Thanks for your tips
Kind regards Ralf

3djc, 2020-03-04:
This message can pop up in two cases : 
- module reports a version less than 1.1.5 (which should not be your case)
- module does not respond properly to more than 2 authentification request

I guess it must be something to do with the second case, but I do not think any of us as such a module. @bsongis  ?

ParkerEde, 2020-03-04:
Now I'm able to reproduce it:
- Power on Horus (Model preset  with ACCESS receiver is activated)
- Press SYS / Tools / start Spectrum analyser 
- RTN after about 10 seconds
- Select another model with D16 receiver
- after seconds the message appears.

This is also happend, if you start with a D16 receiver, start spectrum analyser and change to ACCESS.
And this todo only works after a fresh restart of the Horus X10.

Kind reagrds
Ralf

ParkerEde, 2020-03-04:
...and now I've found a second way to reproduce this issue:

- Power on Horus (Model preset with ACCESS receiver is activated)
- Select another model with D16 receiver
- Change back to the Model preset with ACCESS receiver

Same with other direction from D16 to ACCESS back to D16.

Is this an issue in OpenTX or in firmware from ISRM module?

Thank you.
Ralf

bsongis, 2020-03-06:
We don't send the GET_HARDWARE_INFO_FRAME in the cases described so I guess that the radio receives too many authentication frames (or doesn't reset properly the authentication frames count)

Would you please tell us the version of your module? SYSTEM / MODULE VERSION ?

ParkerEde, 2020-03-06:
[image]

bsongis, 2020-03-06:
Thanks. So definitely not a version issue.

bsongis, 2020-03-06:
From you are saying above, changing from ACCESS to D16 makes the module authenticate again. It doesn't make any sense to me. We only reset the authentication count at module startup, not when there is a parameter change.

I will check on monday with the person in charge of ISRM and get back to you soon.

bsongis, 2020-03-07:
The answer was quick. The module resets when switching from ACCESS to D16 and also when switching from NORMAL mode to SPECTRUM mode.

The original author of the check didn't know about this internal reset at the time it was implemented.

I will write a workaround (authentication counter reset to 0 on each mode switch) on Monday.

3djc, 2020-03-07:
Fix has been done, it will be in next nightly when merged

bsongis, 2020-03-07:
Merged

ParkerEde, 2020-03-28:
Hi Betrand,
now I've updated to 2.3.7 but I still got the message, if I change from model preset with ACCST to model preset with ACCESS and back to model preset with ACCST.

Kind regards
Ralf

PR #7425 (2020-03-07): Reset ISRM upgrade moduel counter reset when mode is changed
This fixes #7422
PR #7511 (2020-04-11): Refactoring to have globals splitted from opentx.h for easier includes
Fixes #7422
closed 2020-04-11 by bsongis