opentx/opentx #7427: override function not returned in lua model.getCustomFunction

lshems, 2020-03-08:
If you have an overrride function set on a custum (special) function, lua willl return a 0 for the function identifier in the model.getCustomFunction().

The model.setCustumFunction() is propably impacted as well.
 

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

kilrah, 2020-03-08:
0 IS the identifier for override.

closed 2020-03-08 by kilrah

lshems, 2020-03-08:
OK, Thanks.

Then I really don't get where I messed up :) creating the logging scripts.

I'll see if I overlooked something. 

Does this mean the channel value is stored in some other parameter?

kilrah, 2020-03-08:
Channel value is stored in value...

lshems, 2020-03-08:
No, No XY problem.

Since the dropdown on Companion is listing the override fucntions in the functionlist, I assumed every overridechX had it's own function ID.

The dropdown on the radio itself actually splits it nicely. function list and channel parameter.

Issue solved (lathough companion is not seperating the entries where the radio does)